gnuradio/pybombs #329: max recursion depth for $variables that cannot be found

gr-sp, 2016-05-12:
I was attempting to build a recipe that had a custom $variable in the configure line of the recipe and I was presented with:
`RuntimeError: maximum recursion depth exceeded while calling a Python object`

After a little sleuthing I determined the cause: `var_replace_all` in recipe.py looks for `n_subs == 0` before it will return. However, for each match of each $variable for which a value cannot be found, that $variable is placed back into the substitution string and n_subs is incremented.

E.g., in the string `"hello $dolly"` if a value cannot be found for `$dolly`, then `$dolly` is placed back into the returned string (by the `var_replace` function) and the returned string is the same as the input string (`"hello $dolly"`). When this happens the return value from `var_re.subn(var_repl, s)` is `("hello $dolly", 1)` even though an actual substitution was not made. I think this is because the count (returned by `subn`) doesn't represent the number of substitutions but the number of times a match occurs; and the number of substitutions is assumed to equal the number of times a match occurs. But since we cannot find a value appropriate for the `$dolly` variable we substitute it with itself. There doesn't appear to be a mechanism for indicating to the `subn` method from the `var_replace` function that a substitution did not actually occur.

So, the end result is that `n_subs == 0` is never `True`.

Ignoring the test for `n_subs == 0` and simply returning `s` from the method seems to work for me. However, I am reluctant to say that this a fix because I assume that someone had a reason for using `re.subn` and then performing a recursion instead of using `re.sub` with no recursion; I just don't know what that reason was.

gr-sp, 2016-05-12:
For the record, the actual configure line in the recipe I was using was:
```
configure: |
  sed -i s@uiPath=@uiPath=\"$prefix\"/www@g ./cfg/cfg.ini
  sed -i s@baseDir=@baseDir=\"$prefix\"@g ./cfg/cfg.ini
  sed -i s@taVersion.*=.*@taVersion\ = \"$ta_version\"@g ./cfg/cfg.ini
  sed -i s@CONFIG_FILE@\"$prefix\"/cfg/cfg.ini@g ./etc/init.d/wifi-gui
  sed -i s@LOG_FILE_PATH@\"$prefix\"/etc/init.d/@g ./etc/init.d/wifi-gui
```

There are two $variables in this line: `$prefix` (x 4) and `$ta_version` (x 1). `var_replace_all` was in an infinite loop attempting to find/replace `$ta_version`. All `$prefix` substitutions were made successfully.

mbr0wn, 2016-05-12:
ermagerd, I broke that in a recent commit. Thanks for catching.

mbr0wn, 2016-05-13:
This should be fixed on master. Do you mind testing for us? Thanks!

gr-sp, 2016-05-13:
I'll check it out sometime in the next few days...no problem. At a glance it looks like it would work.

However, I would like to suggest that this is an "all stop" error. If there is a replacement variable in a recipe that can't be found, then I would argue that a missing configuration variable is a major problem. Of course, it totally depends on the specific variable in question as to the level of impact. But, let's assume that it's possible that somehow "prefix" is missing from the configuration (although this is extremely unlikely). What kind of impact would it have on the build if every instance of `$prefix` in every recipe was replaced with an empty string. To top it off, it would happen silently.

I am thinking this would be a disaster for the user and the error they get may not even lead them to the conclusion that the "prefix" configuration value is missing. Obviously this is an extreme case, but I could imagine that someone would use a configuration variable liberally throughout a set of recipes and that configuration variable could be quite important for their use case. If it is a path (or part of a path) to a directory or a file it could be very bad.

mbr0wn, 2016-05-13:
That's a good point, but a good case can also be made for keeping the strings empty. We could easily show a warning, and maybe that's the "best of both worlds" solution. The reason for replacing with empty strings is that we run commands in a shell-like fashion, and there, it's common to do the null-replace thing.

I'm not convinced by the current status quo, though. Just halting might be safer. We can always fix recipes, if need be, and certain variables are always supposed to be there (like `$prefix` and `$makewidth`), so we might as well halt on them.

closed 2016-05-14 by mbr0wn

gr-sp, 2016-05-16:
OK...I checked on this one. It breaks out with a nice easy to read message:

"Could not expand variable $ta_version."

Thanks!