gnuradio/pybombs #330: config options in prefix local config file are ignored

gr-sp, 2016-05-12:
During my adventures with #329, another issue came up.

#329 refers to the `$ta_version` variable that is resulting in an infinite loop in an attempt to replace `$ta_version` in a recipe with a specific value.

I updated my prefix local configuration file (<prefix>/.pybombs/config.yml) with the following:
```
config:
  ta_version=0.0.0.0
```
However, `var_replace_all` and by association `var_replace` could not find `ta_version` in the config. So, I updated my user local configuration file ($HOME/.pybombs/config.yml) with the same as above and the `ta_version` key was found and replaced in the recipe with no problem.

So, I am not sure of the author's intent, but I though that each of these configuration files could contain the same items but that the items in the prefix local configuration file would override the same items in other configuration files. However, it appears that configuration items in the prefix local configuration file are being ignored.

Also, I tried to update both of my configuration files independently with the following:
```
vars:
  ta_version=0.0.0.0
```
and `$ta_version` still failed to be found. Was I doing this right? Is it not supposed to work this way?

mbr0wn, 2016-05-13:
That is, indeed, the correct intent and I know it was working at some point in the past.

mbr0wn, 2016-05-14:
Could reproduce. Found a bug in config_manager.py. Submitted fix. Tested locally. Would appreciate some confirmation from you!

closed 2016-05-14 by mbr0wn

gr-sp, 2016-05-16:
So nice. My config variable in my prefix local config now overrides the one in my user local config.

However, I have a question.

I tried to put my config variable (`$ta_version`) in my config file as:
```
vars:
  ta_version: X.X.X.X
```
and that didn't work. So I tried again with:
```
vars:
  config_opts:
    ta_version: X.X.X.X
```
and that also didn't work. The only one that seems to work is:
```
config:
  ta_version: X.X.X.X
```
Am I misunderstanding how the `vars:` config section works?

mbr0wn, 2016-05-16:
I think you are misunderstanding it. The 'vars' section is per-recipe, so this would work:

```{.yaml}
packages:
  pkgname:
    vars:
     ta_version: x.x.x.x
```