gnuradio/pybombs #333: Sometimes default [Y]/N crashes PyBOMBS

mhostetter, 2016-05-15:
I have had this happen several times and on different machines.  When running a PyBOMBS command that requires a confirmation, just hitting enter and not explicitly selecting yes or no can cause a crash.  I have found explicitly selecting an option will allow it to work.

```
matt@t430:~$ pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
Alias `gr-recipes' already exists, overwrite Y/[N]? 
PyBombs.recipes - WARNING - Aborting.
matt@t430:~$ pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
Alias `gr-recipes' already exists, overwrite Y/[N]? Y
PyBombs.recipes - WARNING - Cache dir /home/matt/.pybombs/recipes/gr-recipes for remote recipe location gr-recipes already exists! Deleting.
Cloning:     (100%) [=========================================================]
matt@t430:~$ 
```

I'll keep you posted if I find more instances of this.

mbr0wn, 2016-05-15:
@mhostetter I guess this is on Python 2?

mbr0wn, 2016-05-15:
In the example you cite, aborting is the correct behaviour. It asks you if you want to overwrite -- if you say 'no', which is the default, then all it can do is quit. The output is not very useful, I'll admit that. But it's not a bug.

Do you have other instances where it actually crashes? Or is this the only case where it fails?

mhostetter, 2016-05-15:
Oh, wow...  Well that's what happens when you pretend to be lucid at 2:30 am.  There have been other times, but I'm curious if it was this same kind of thing.  Maybe a false alarm.

mbr0wn, 2016-06-18:
Still don't think this is a bug. Will reopen on more evidence.

closed 2016-06-18 by mbr0wn