gnuradio/pybombs #335: Character encoding conflict with progress bar.

gr-sp, 2016-05-16:
During my build I received the output that follows. When I perform the build a second time, it doesn't throw this error. It looks like there might be an issue with one of the recent commits that deals with character encoding. Another thing that is odd about this is that although an error is thrown in what appears to be the "building" state, pybombs continues on into the "installing" state.

FYI, I am using a pyboms prefix that has a Python 2.7.11 virtualenv on CentOS 6.5.

```
PyBombs.install_manager - INFO - Installing package: sip
PyBombs.PackageManager - DEBUG - install(sip, static=False)
PyBombs.PackageManager - DEBUG - Package sip is requesting a source build.
PyBombs.PackageManager - DEBUG - Using packager source
PyBombs.Packager.source - DEBUG - State on package sip is 0
PyBombs.Fetcher - DEBUG - Fetching source for recipe: sip
PyBombs.Fetcher - DEBUG - Switching to src directory: /home/shawn/prefixes/ta2/src
PyBombs.Fetcher - DEBUG - Getting fetcher for http://pkgs.fedoraproject.org/repo/pkgs/sip/sip-4.18.tar.gz/78724bf2a79878201c3bc81a1d8248ea/sip-4.18.tar.gz, type wget
PyBombs.Fetcher - DEBUG - Using fetcher <pybombs.fetchers.wget.Wget object at 0x7f357f1dd690>
PyBombs.Fetcher.wget - DEBUG - Requiring packages on host system: ['python-requests']
PyBombs.Fetcher.wget - DEBUG - Requirements met.
00992 kB / 00992 kB (100%)
PyBombs.extract_to - DEBUG - Unpacking sip-4.18.tar.gz
PyBombs.extract_to - DEBUG - Common prefix: sip-4.18
PyBombs.extract_to - DEBUG - Moving /tmp/tmpXvoSJ_/sip-4.18 -> sip
PyBombs.Inventory - DEBUG - Setting state to `fetched'
PyBombs.Inventory - DEBUG - Saving inventory to file /home/shawn/prefixes/ta2/.pybombs/inventory.yml...
PyBombs.Packager.source - DEBUG - Using build directory: /home/shawn/prefixes/ta2/src/sip
PyBombs.Packager.source - DEBUG - Configuring recipe sip
PyBombs.Packager.source - DEBUG - Using vars - {'config_opt': '', 'builddir': '/home/shawn/prefixes/ta2/src/sip'}
PyBombs.Packager.source - DEBUG - In cwd - /home/shawn/prefixes/ta2/src/sip
PyBombs._process_thread() - DEBUG - Executing command `python configure.py -b /home/shawn/prefixes/ta2/bin -e /home/shawn/prefixes/ta2/include/ -d /home/shawn/prefixes/ta2/lib/python2.7/site-packages/ -v /home/shawn/prefixes/ta2/share/sip/'
Configuring: (100%) [======================================================================================================================================================================================================================]
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 0
PyBombs.Packager.source - DEBUG - Configure successful.
PyBombs.Inventory - DEBUG - Setting state to `configured'
PyBombs.Inventory - DEBUG - Saving inventory to file /home/shawn/prefixes/ta2/.pybombs/inventory.yml...
PyBombs.Packager.source - DEBUG - Building recipe sip
PyBombs.Packager.source - DEBUG - In cwd - /home/shawn/prefixes/ta2/src/sip
PyBombs._process_thread() - DEBUG - Executing command `make -j4'
Building:    [\                                                                                                                                                                                                                            ]Exception in thread Thread-15:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/utils/subproc.py", line 150, in _process_thread
    ret_code = run_with_output_processing(proc, o_proc, event, kwargs.get('cleanup'))
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/utils/subproc.py", line 112, in run_with_output_processing
    o_proc.process_output(line_stdout, line_stderr)
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/utils/output_proc.py", line 101, in process_output
    self._check_for_percentage(stdoutdata)
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/utils/output_proc.py", line 106, in _check_for_percentage
    if self.percent_regex.search(str(strdata)) is not None:
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2018' in position 20: ordinal not in range(128)

PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 0
PyBombs.Packager.source - DEBUG - Make successful
PyBombs.Inventory - DEBUG - Setting state to `built'
PyBombs.Inventory - DEBUG - Saving inventory to file /home/shawn/prefixes/ta2/.pybombs/inventory.yml...
PyBombs.Packager.source - DEBUG - Installing package sip
PyBombs.Packager.source - DEBUG - In cwd - /home/shawn/prefixes/ta2/src/sip
PyBombs._process_thread() - DEBUG - Executing command `make install'
Installing:  (100%) [======================================================================================================================================================================================================================]
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 2
PyBombs.Packager.source - ERROR - Problem occurred while building package sip:
Installation failed
PyBombs.install_manager - ERROR - Error installing package sip. Aborting.
```

gr-sp, 2016-05-16:
This error is not limited to the package shown (sip) in the output above. I just got the same error thrown while attempting to build pyqt4.

gr-sp, 2016-05-16:
It appears that a workaround for this issue may be to use `pybombs -vv install <pkg>`. Although...uhg...I am now spammed with the low level build output.

gr-sp, 2016-05-16:
I got the same error while attempting to build pyqwt5. I think the pattern here is that all of these errors are thrown from source builds of python packages. However, the pyqwt5 build did not halt pybombs build progress. It continued on to the next package (this leaves me a little unconvinced that pyqwt5 was built/installed properly).

stwunsch, 2016-05-16:
that's most likely my fault due to the ugly string cast. this patch should fix it for you:

```
commit 109885de509e43ffc9d6d5f3ae917e9c0e261101
Author: Stefan Wunsch <stefan.wunsch@student.kit.edu>
Date:   Mon May 16 17:38:50 2016 +0200

    fix encoding problem, revert py3 changes

diff --git a/pybombs/utils/output_proc.py b/pybombs/utils/output_proc.py
index ad714bb..2662908 100644
--- a/pybombs/utils/output_proc.py
+++ b/pybombs/utils/output_proc.py
@@ -103,7 +103,7 @@ class OutputProcessorMake(OutputProcessor):
 
     def _check_for_percentage(self, strdata):
         " Check if strdata contains a percentage string. "
-        if self.percent_regex.search(str(strdata)) is not None:
+        if self.percent_regex.search(strdata) is not None:
             self.percent_found = True
 
     def _update_percentage(self, strdata):
```

In words: Just remove the `str(...)`.

gr-sp, 2016-05-16:
Nicely done...this works.

stwunsch, 2016-05-16:
@mbr0wn you should patch this in master as well. i am not sure whether a strdata.decode() solves the problem nicely or it should be fixed earlier during the output handling.

mbr0wn, 2016-05-16:
@gr-sp, @stwunsch I've seen this too and added 838309e29525725f4c046da59db130e47dda7d34, which I thought fixes this. Are you running a PyBOMBS later than this commit?

gr-sp, 2016-05-16:
Yes, as of first thing this morning I was using https://github.com/gnuradio/pybombs/commit/a698320731c7021033234beba38cc5d2cdf4edcd and the problem exists in this commit. I manually applied the diff provided which fixed the problem.

gr-sp, 2016-05-16:
I just updated to the latest commit (12a8605442246ed1004f6083f13a129dd0b0a6f8) and the problem has been fixed for me.

closed 2016-05-16 by mbr0wn