gnuradio/pybombs #336: Packages removed in wrong order.

gr-sp, 2016-05-16:
Lets say I have 3 recipes that are built from source (pkg1.lwr, pkg2.lwr, and pkg3.lwr). Then, lets say that pkg3 depends on pkg2 which depends on pkg1.

So, `pybombs install pkg3` installs pkg1, then pkg2, and finally pkg3.

However, `pybombs remove pkg1` results in the removal of pkg1, then pkg2, and finally pkg3.

The problem with this is that with real recipes sometimes the uninstall will fail if dependencies are missing. I have noticed this specifically with cmake projects although I cannot say if this is the only type of project that would complain. Also, I do not have the specific output that is generated in front of me at the moment. However, continuing with our made up recipes from above, pkg2's uninstall step could fail because pkg1 has already been removed. The removal should occur in the reverse order: remove pkg3, then pkg2, then finally pkg1.

I think this used to work correctly.

This is easily tested with fake recipes that simply echo statements to the screen for their configure, make, install, and uninstall steps.

mbr0wn, 2016-05-16:
Good catch. I would like to add a `serialize()` call to a dep_tree object that returns the tree in serialized fashion, which means if you *install* these packages *in order*, you will have the dependencies properly resolved. Applying a `reverse()` to a serialized dep tree should then do the trick for removing.

mbr0wn, 2016-05-16:
...added the serialize() call, but need to pull it into remove.

mbr0wn, 2016-05-17:
Should be good now. Appreciate comments.

gr-sp, 2016-05-17:
This checks out on my system...thanks!

closed 2016-05-17 by mbr0wn