gnuradio/pybombs #338: setuptools invalid command name

gr-sp, 2016-05-17:
I keep having to remove the install_opt var from setuptools.lwt. When building I get:
```
PyBombs._process_thread() - DEBUG - Executing command `python setup.py install --prefix=/home/shawn/prefixes/ta2 setuptools.satisfier==deb?{--install-layout=deb}:{}'
invalid command name 'setuptools.satisfier==deb?{--install-layout=deb}:{}'
```

gr-sp, 2016-05-17:
OK...so I can't remove it anymore because then $install_opt isn't defined (which is the way it should be). But I can change it from:
```
vars:
        install_opt: "setuptools.satisfier==deb?{--install-layout=deb}:{}"
```
to
```
vars:
        install_opt:
```

mbr0wn, 2016-05-18:
Yeah that's stupid. PB1 had a magic way of being conditional inside variables. Until we have something like that, it's empty.

closed 2016-05-18 by mbr0wn