gnuradio/pybombs #339: gqrx.lwr breaks gnuradio source build

mhostetter, 2016-05-18:
I installed gnuradio through PyBOMBS.  Later I ran `$ pybombs install gqrx`.  It seems gqrx wants to install gnuradio deb packages.  Either way, neither `gnuradio-companion` nor `gqrx` run afterward.

```
matt@t430:~$ gnuradio-companion
Traceback (most recent call last):
  File "/home/matt/pybombs/target/bin/gnuradio-companion", line 29, in <module>
    from gnuradio.grc.main import main
ImportError: No module named main
```

```
matt@t430:~$ gqrx
linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.git-60-g9e0861e1

gr-osmosdr 0.1.1 (0.1.1) gnuradio 3.7.2.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace 
gqrx: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgnuradio-osmosdr.so.0.0.0: undefined symbol: _ZN3uhd6device4findERKNS_13device_addr_tE
```

Then when I try and remove gqrx, I get this error.  :( Sad days...
```
matt@t430:~$ pybombs remove gqrx
PyBombs.remove - ERROR - Package gqrx is not installed. Aborting.
matt@t430:~$ pybombs -v remove gqrx
PyBombs.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {}
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/matt/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {'default_prefix': 'myprefix'}
PyBombs.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/templates
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/matt/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Using environment variable PYBOMBS_PREFIX as prefix (/home/matt/pybombs/target)
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/matt/pybombs/target/.pybombs
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/matt/pybombs/target/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/matt/pybombs/target/src
PyBombs.Inventory - DEBUG - Trying to load inventory file /home/matt/pybombs/target/.pybombs/inventory.yml...
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/matt/pybombs/target/.pybombs/recipes
PyBombs.ConfigManager - DEBUG - Full list of recipe locations: ['/home/matt/pybombs/target/.pybombs/recipes', '/home/matt/.pybombs/recipes/gr-etcetera', '/home/matt/.pybombs/recipes/gr-recipes', '/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes']
PyBombs.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/matt/pybombs/target/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/matt/pybombs/target/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/matt/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/matt/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 51 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/matt/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/matt/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 165 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 6 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.remove - DEBUG - Initializing command class for command remove
PyBombs.PackageManager - DEBUG - Using packagers: ['source']
PyBombs.remove - DEBUG - Resolving dependency list for clean removal.
PyBombs.PackageManager - DEBUG - Checking if package gqrx is installed.
PyBombs.remove - ERROR - Package gqrx is not installed. Aborting.
matt@t430:~$
```

For completeness, this is the output from installing gqrx.
```
matt@t430:~$ pybombs install gqrx
|
\- pulseaudio
|
\- gqrx
[... 331 lines cut ...]
Setting up libgnuradio-wxgui3.7.2.1:amd64 (3.7.2.1-5ubuntu0.1) ...
Setting up gr-iqbal (0.37.1.5.d4fd4d-1) ...
Setting up python-opengl (3.0.2-1) ...
Setting up gnuradio (3.7.2.1-5ubuntu0.1) ...
Begin freedesktop install...
Install icon: 32x32
Install icon: 48x48
Install icon: 64x64
Install icon: 128x128
Install icon: 256x256
Install mime type
Install menu items
Done!
Setting up gqrx-sdr (2.2.0.74.d97bd7-2) ...
Setting up gr-fcdproplus (0.0.1.1.2c80be-3) ...
Setting up qthid-fcd-controller (4.1-3) ...
Setting up uhd-host (3.5.5-1) ...
Adding group `usrp' (GID 126) ...
Done.
net.core.rmem_max = 50000000
net.core.wmem_max = 1048576
Processing triggers for libc-bin (2.19-0ubuntu6.7) ...
PyBombs.install - INFO - Installation successful.
matt@t430:~$ 
```

mbr0wn, 2016-05-18:
Hmmm. I pushed quick patches to pybombs and gr-recipes to avoid this. But what's the correct fix?

It *should* be doing something like checking if dependencies are already built from source in the current prefix, and the building all dependees from source, too.

mhostetter, 2016-05-18:
I previously had PyBOMBS installed through pip.  So I `$ sudo pip uninstalled pybombs` and then built the latest repo from source using `$ python setup.py build` and `$ sudo python setup.py install`.  I also ran `$ pybombs recipes update gr-recipes`.

This is what I get now when trying to remove gqrx.
```
matt@t430:~/pybombs$ pybombs install gqrx
PyBombs.install_manager - INFO - No packages to install.
matt@t430:~/pybombs$ pybombs -v remove gqrx
PyBombs.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {}
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/matt/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {'default_prefix': 'myprefix'}
PyBombs.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/templates
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/matt/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Using environment variable PYBOMBS_PREFIX as prefix (/home/matt/pybombs/target)
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/matt/pybombs/target/.pybombs
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/matt/pybombs/target/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/matt/pybombs/target/src
PyBombs.Inventory - DEBUG - Trying to load inventory file /home/matt/pybombs/target/.pybombs/inventory.yml...
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/matt/pybombs/target/.pybombs/recipes
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/matt/pybombs/target/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {}
PyBombs.ConfigManager - DEBUG - Full list of recipe locations: ['/home/matt/pybombs/target/.pybombs/recipes', '/home/matt/.pybombs/recipes/gr-etcetera', '/home/matt/.pybombs/recipes/gr-recipes', '/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes']
PyBombs.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/matt/pybombs/target/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/matt/pybombs/target/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/matt/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/matt/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 51 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/matt/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/matt/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 165 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 11 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.remove - DEBUG - Initializing command class for command remove
PyBombs.PackageManager - DEBUG - Using packagers: ['source']
PyBombs.remove - DEBUG - Resolving dependency list for clean removal.
['gqrx']
PyBombs.PackageManager - DEBUG - Checking if package gqrx is installed.
PyBombs.remove - ERROR - Package gqrx is not installed. Aborting.
matt@t430:~/pybombs$ 
```

mhostetter, 2016-05-18:
Any suggestions on how to remove gqrx gracefully?  I had to re-image the last time this happened.  :(

mbr0wn, 2016-05-18:
gqrx is just a binary. Find all files called 'gqrx' and delete them.

mhostetter, 2016-05-18:
I guess I was more referring to uninstalling all the dependencies/gnuradio binaries that got installed by PyBOMBS when I ran `$ pybombs install gqrx`.  Those binaries are causing conflicts with the gnuradio source build.

mbr0wn, 2016-05-18:
You'll have to apt-get remove them, I guess.

mbr0wn, 2016-05-21:
F'up in #342.

closed 2016-05-21 by mbr0wn