gnuradio/pybombs #340: Dependency check not working.

gr-sp, 2016-05-18:
I have a recipe (gcc.lwr) which is depending on another recipe (glibc-devel.i686.lwr) but the dependency check isn't working.

The first problem is that pybombs is attempting to install the `glibc-devel.i686` package even though it is already installed (I verified it is installed with `rpm -qa | grep glibc`). In the full output below, you will find the following two lines that show it attempted to install an already installed package:
```
Package glibc-devel-2.12-1.166.el6_7.7.i686 already installed and latest version
Nothing to do
```

The second problem is that the error that is thrown isn't clear. The last few lines of the below output are:
```
PyBombs.PackageManager - DEBUG - Using packager pkgconfig
PyBombs.Packager.pkgconfig - OBNOXIOUS - install(glibc-devel.i686, static=False)
PyBombs.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBombs.PackageManager - DEBUG - Using packager cmd
PyBombs.Packager.cmd - OBNOXIOUS - install(glibc-devel.i686, static=False)
PyBombs.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBombs.PackageManager - DEBUG - Using packager source
PyBombs.Packager.source - WARNING - Cannot find a source URI for package glibc-devel.i686
PyBombs.install_manager - ERROR - Error installing package glibc-devel.i686. Aborting.
```
This indicates a WARNING that the source URI cannot be found and then immediately aborts. There is no source line in the recipe I am using so the WARNING makes sense. But why is it even checking for a source location? It just verified that the package was installed. Why is it aborting? There is no message that specifically states the problem.

It seems like it isn't properly detecting the installed glibc-devel.i686 package and that, all evidence to the contrary, it is determined to insist that the package needs to be installed.

All that being said, I ended up in this mess because my glibc recipe incorrectly specified `rpm: glibc-devel` instead of `rpm: glibc-devel.i686`. My gcc build failed (cannot find stubs-32.h) and I realized the mistake. So, I updated glibc-devel.i686.lwr to have the correct rpm value (`rpm: glibc-devel.i686`) and ran `pybombs -v install gcc` again and got the following output.
```
[shawn@cos65shawn ta-recipes]$ pybombs -v install gcc
PyBombs.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {}
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/shawn/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {'default_prefix': 'ta'}
PyBombs.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/templates
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/shawn/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias ta.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/home/shawn/prefixes/truearrow)
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/shawn/prefixes/truearrow/.pybombs
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/shawn/prefixes/truearrow/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/shawn/prefixes/truearrow/src
PyBombs.Inventory - DEBUG - Trying to load inventory file /home/shawn/prefixes/truearrow/.pybombs/inventory.yml...
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/shawn/prefixes/truearrow/.pybombs/recipes
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/shawn/prefixes/truearrow/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {'ta_version': '6.0.0.0'}
PyBombs.ConfigManager - DEBUG - Full list of recipe locations: ['/home/shawn/prefixes/truearrow/.pybombs/recipes', '/home/shawn/.pybombs/recipes/ta-recipes', '/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes']
PyBombs.ConfigManager - DEBUG - Named recipe locations: {'ta-recipes': 'git+spenny@10.0.0.97:/var/gitroot/ta-recipes.git'}
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/shawn/prefixes/truearrow/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/shawn/prefixes/truearrow/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/shawn/.pybombs/recipes/ta-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/shawn/.pybombs/recipes/ta-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 181 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 11 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.install - DEBUG - Initializing command class for command install
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBombs.PackageManager - DEBUG - yumdnf is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager port
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBombs.PackageManager - DEBUG - pkgconfig is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['pip', 'yumdnf', 'pkgconfig', 'cmd', 'source']
[... 55 lines cut ...]
Loaded plugins: fastestmirror, refresh-packagekit, security
Loading mirror speeds from cached hostfile
 * base: mirror.us.leaseweb.net
 * extras: mirrors.advancedhosters.com
 * updates: mirror.dattobackup.com
Setting up Install Process
Package glibc-devel-2.12-1.166.el6_7.7.i686 already installed and latest version
Nothing to do
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 0
PyBombs.PackageManager - DEBUG - Using packager pkgconfig
PyBombs.PackageManager - DEBUG - Using packager cmd
PyBombs.PackageManager - DEBUG - Using packager source
PyBombs.Packager.source - WARNING - Cannot find a source URI for package glibc-devel.i686
PyBombs.install_manager - ERROR - Error installing package glibc-devel.i686. Aborting.
[shawn@cos65shawn ta-recipes]$ rpm -qa | grep glibc
glibc-headers-2.12-1.166.el6_7.7.x86_64
glibc-devel-2.12-1.166.el6_7.7.i686
glibc-2.12-1.166.el6_7.7.x86_64
glibc-2.12-1.166.el6_7.7.i686
glibc-common-2.12-1.166.el6_7.7.x86_64
glibc-devel-2.12-1.166.el6_7.7.x86_64
```

This may just be some weird issue that is a result of the order of operations that I performed, however, it still seems like pybombs is not doing what it should. This may be difficult to reproduce so I am posting this issue with lots of background so I don't lose the history.

gr-sp, 2016-05-18:
`pybombs inv` shows:
```
Showing package state:
gcc:	configured
autoconf:	installed
mpc:	installed
apache-thrift:	fetched
cloog:	installed
isl:	installed
gmp:	installed
automake:	installed
binutils:	installed
mpfr:	installed
```

`pybombs recipes list` shows:
gcc    <recipe location>    -
glibc-devel.i686    <recipe location>    yumdnf

mbr0wn, 2016-05-18:
Can you please post the contents of those recipes, too?

mbr0wn, 2016-05-18:
I have a hunch that the . in the recipe name is tripping up our regexes.

gr-sp, 2016-05-18:
I changed the name of the recipe back to `glibc-devel-i686.lwr` (this was my original name for the recipe) and updated the `gcc.lwr` recipe to depend on the new recipe name and it now seems to be working correctly.

I think your right about the recipe name regexes being a little thrown off by `glibc-devel.i686.lwr`.

Aw man...I think I remember what happened. After I got the gcc build error (missing stubs-32.h) I had to look to make sure that the `glibc-devel-i686.lwr` recipe existed. It did so, for some reason that I cannot remember now, I changed the name of the recipe to `glibc-devel.i686.lwr`. I didn't initially notice that `rpm: glibc-devel` wasn't correct in the recipe. So that kind of got me using the `glibc-devel.i686.lwr` recipe. I eventually noticed the rpm name wasn't correct in the recipe, so I updated it without changing the name of the recipe back to its original name and then ended up with this problem.

Well, I supposed this could still be an issue but if you keep this issue I would recommend change the name of it: "gr-sp can't debug recipe problems". Just kidding...maybe something more like: "Recipe file name formats are limited".

gr-sp, 2016-05-18:
At least you have some history showing what happens if pybombs doesn't like your recipe name!

gr-sp, 2016-05-18:
Possibly spoke too soon...standby...

mbr0wn, 2016-05-18:
Actually, I think you *can* have a . in your recipe name. But not in your rpm: line.
You *should* be able to specify an arch in a RPM based recipe. And that's when you need the dot. I've added a quick hack, but won't be able to give it much of a testing.

mbr0wn, 2016-05-18:
See this line from your log:

    PyBombs.Packager.yumdnf - DEBUG - Package glibc-devel.i686 has version 2.12 in yum

It means its handling your recipe correctly, at least up till then.

mbr0wn, 2016-05-18:
Try 4bd6658 with your old recipe. Note this is blind fly-by patching.

gr-sp, 2016-05-18:
K...standby

gr-sp, 2016-05-18:
Yup...works here.

Holy crap that sucked.

mbr0wn, 2016-05-18:
So... it works, or it sucks, or both?

gr-sp, 2016-05-18:
I verified that it works with either recipe name (both `glibc-devel-i686.lwr` and `glibc-devel.i686.lwr`) when the rpm package name is specified as `rpm: glibc-devel.i686`.

Both!

mbr0wn, 2016-05-18:
Cool, thanks. Closing this. Thanks for testing real-time chat in github, and testing this patch.

Let's continue making stuff suck less :)

closed 2016-05-18 by mbr0wn

gr-sp, 2016-05-18:
Agreed.