gnuradio/pybombs #342: Switch to source packager if dependencies were built from source

mbr0wn, 2016-05-20:
Say you have a dep tree as such:
```
pkg1
\-+ pkg2
  |
  \- pkg3
```
Now imagine you've installed pkg3 from source. The only correct way to continue is to also build pkg2 and pkg1 from source, too. 

Currently, PyBOMBS doesn't do that. It'll build pkg3 from source, then the next leaf node (pkg2) has its completely own logic about choosing source or not. Same for pkg1.

This is why @mhostetter ran into #339.

The current behaviour is buggy, but this will require some more changes than a small bugfix, I guess.

NinjaComics, 2016-05-21:
@mbr0wn Correct me if I am wrong, but shouldn't the approach to chose the packager be independent for each package ? Say I have ten packages under the main package to be installed, and if the tenth package is built from source, I will have to wait for a long time (depending upon the type of packages) to install the other 9 packages also from source, which is a time consuming process (building anything from source is always a tedious process ?).

mbr0wn, 2016-05-23:
The fact that it takes time to build from source is inconsequential -- if you want a source build, it'll just take time. The idea of PyBOMBS is that it doesn't require handholding, so you can just kick off the build before you go to bed if you like.

Now, for the choice of packager, what you suggest is what we currently have. However, it doesn't really make sense. Say you request a source build of UHD, but for GNU Radio, you say 'pick whatever'. PyBOMBS will first install UHD from source. So far, so good. Now, in this prefix, you would expect other packages to use the UHD version you just built from source, right? However, in this case, PyBOMBS might decide to install GNU Radio via apt-get. This will pull in another, different UHD version through apt-get also (and thus pollute your system).
Same goes for any other library -- if you've went out of your way to tell PyBOMBS to build it from source, you want to *use* it. The only way to use it is to also build dependees from source.