gnuradio/pybombs #346: Recipe resolution order

gr-sp, 2016-05-31:
Let's say I have added a set of user-specific recipes:
`pybombs recipes add  first_recipes <location of recipes 1>`
Then, I add a set of prefix-specific recipes like so:
`pybombs -p <my prefix> recipes add second_recipes <location of recipes 2>`

Now, let's say that both recipe sets have the recipe `test.lwr` in them. When I execute `pybombs -v install test`, the `test.lwr` from the `first_recipes` set is invoked instead of the `test.lwr` from the `second_recipes` set.

The reason this is a problem is because the `first_recipes` set is in `$HOME/.pybombs/recipes/first_recipes` and is common to all prefixes for a given user. The `second_recipes` set is in `<my prefix>/.pybombs/recipes/second_recipes` and is specific to `<my prefix>`. The prefix-specific recipes cannot be used to override the user-specific recipes.

If you happen to drop the `test.lwr` from `<my_prefix>/.pybombs/recipes/second_recipes` into `<my prefix>/.pybombs/recipes` then you will get the correct resolution order.

mbr0wn, 2016-05-31:
This is not how it should be. Lemme check.

gr-sp, 2016-06-01:
Hmm...I just realized that the recipes in the `second_recipes` set are not getting picked up at all.

Originally I was just trying to override recipes in `first_recipes` so it never occurred to me that none of the recipes in `second_recipes` are being picked up.

To recreate, follow the above instructions to create the `second_recipes` set. Then, just put a unique recipe (something like `test2.lwr`) in `<my prefix>/.pybombs/recipes/second_recipes` and attempt to install it (`pybombs install test2`). You will most likely get something along the lines of the following:
```
PyBombs.get_recipe - ERROR - Error fetching recipe `test2':
Package test2 has no recipe file!
```

mbr0wn, 2016-06-08:
Found something dumb and obvious and fixed in f23a35a19753485db36080765c72588436698903. Still needs some testing, and `pybombs prefix list-repos` is not printing the correct order, but I think this might help your specific issue.

mbr0wn, 2016-06-08:
@gr-sp btw, any more bugs incoming? I'd like to tag another release :)

gr-sp, 2016-06-08:
This does seem to work. I can put a `test.lwr` in `<prefix>/.pybombs/recipes/test-recipes/` and successfully override a `test.lwr` in `$HOME/.pybombs/recipes/gr-recipes/`.

gr-sp, 2016-06-08:
I put in issue #351 and I have one more issue to submit related to regexes I think.

mbr0wn, 2016-06-09:
b41c9e721daf178ecf21bf5c2f8a71c7073be131

closed 2016-06-09 by mbr0wn