gnuradio/pybombs #348: List out info about all the prefixes

NinjaComics, 2016-06-03:
For the idea described in #345 (refer screenshot below), it is required to get the list of prefix aliases the user has created earlier. The idea is to add a subcommand listing out info about all the prefixes.     

[image]

mbr0wn, 2016-06-03:
I would suggest a sub-sub-command for `pybombs prefix`.

NinjaComics, 2016-06-03:
There's already a `pybombs prefix info` which lists out the current prefix directory. I was thinking if we can modify it to list out all the prefix(with aliases) and highlight the active prefix.

mbr0wn, 2016-06-03:
Yeah that's a good place. `prefix info` is currently not all that useful.

mbr0wn, 2016-06-18:
22ef131134621e2b936d5f07249b6ef152ebc29f

closed 2016-06-18 by mbr0wn