gnuradio/pybombs #349: pybombs doesn't install 'libcomedi-dev' automatically

scateu, 2016-06-03:
which will led to 

```
[ 67%] Building CXX object gr-comedi/lib/CMakeFiles/gnuradio-comedi.dir/sink_s_impl.cc.o
In file included from /home/scateu/gnuradio/src/gnuradio/gr-comedi/lib/sink_s_impl.cc:29:0:
/home/scateu/gnuradio/src/gnuradio/gr-comedi/lib/sink_s_impl.h:28:23: fatal error: comedilib.h: No such file or directory
 #include <comedilib.h>
                       ^
compilation terminated.
make[2]: *** [gr-comedi/lib/CMakeFiles/gnuradio-comedi.dir/sink_s_impl.cc.o] Error 1
make[1]: *** [gr-comedi/lib/CMakeFiles/gnuradio-comedi.dir/all] Error 2
make: *** [all] Error 2
PyBombs.Packager.source - ERROR - Build failed. See output above for error messages.
PyBombs.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBombs.install - ERROR - Error installing package gnuradio. Aborting.
```

## My system info

```
$ pybombs --version
2.0.1

$ uname -a
Linux scateu-ThinkPad-X230 3.19.0-49-generic #55~14.04.1-Ubuntu SMP Fri Jan 22 11:24:31 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

mbr0wn, 2016-06-03:
What did you do to trigger this? GNU Radio won't error if it can't find libcomedi headers; it simply won't build that component.

mbr0wn, 2016-06-10:
Closing this due. Don't think this is a PyBOMBS issue, anyway.

closed 2016-06-10 by mbr0wn