gnuradio/pybombs #350: prefix not correct

gr-sp, 2016-06-06:
After commit ebcf3a52d62840ba47d8155466224c107d02903f, I have been experiencing some odd prefix behavior.

Before the commit, a `pybombs prefix info` prints the correct prefix directory.

After the commit, a `pybombs prefix info` prints the incorrect prefix directory. The printed directory seems to be dependent on the current working directory. Also it is incorporating the prefix alias into the prefix path. For example, suppose I initialize a new prefix

- `cd <prefix_parent>`
- `pybombs prefix init -a testalias test`

This creates a new prefix directory `test` in the `<prefix_parent>` directory with an alias `testalias`. Now, performing a `pybombs prefix info` results in `Prefix dir: <prefix_parent>/testalias`. Not correct. It should be `Prefix dir: <prefix_parent>/test`. Now, working from `<prefix_parent>` perform `cd` to `<prefix_parent_parent>`.

- `cd ..`

Performing `pybombs prefix info` from the parent directory of `<prefix_parent>` (AKA `<prefix_parent_parent>`) results in `Prefix dir: <prefix_parent_parent>/testalias`. This is also not correct. It is incorporating the current working directory, in this case `<prefix_parent_parent>`, into the prefix path and then appending the prefix alias `testalias` to the end of the path.

Note: I did not find this issue by performing `pybombs prefix info`. I was just having some problems attempting to fetch the source of some recipes. Verbose output indicated that there was something fishy going on with the prefix. However, detailing that process is too much to put in an issue. Hopefully this will be enough to solve the problem. If not, I will go into more grueling details.

gr-sp, 2016-06-07:
Is it possible that commit https://github.com/gnuradio/pybombs/commit/ebcf3a52d62840ba47d8155466224c107d02903f can be reverted?

mbr0wn, 2016-06-07:
Note that this:

    $ pybombs prefix init -a testalias test

is not possible without ebcf3a5, at least in some configurations. If you have relative paths to prefixes, it can screw up certain operations. So no, it can't be reverted, but it seems there's some other bug here that needs fixing.

gr-sp, 2016-06-07:
I have always init'd my prefixes in this manner. Are you saying that my prefixes shouldn't have worked before?

mbr0wn, 2016-06-07:
If it works, it works. If not, it fails quite obviously. The reason I added this patch was because this worked fine (abs path provided on cmd line):

    $ pybombs prefix init -R gnuradio-default ~/prefix/foo

But not this:

    $ cd ~/prefix
    $ pybombs prefix init -R gnuradio-default ./foo

I'm surprised it's so different for you. In any case, if I revert this, I'm still left with a bug. I'll try and catch your bug, and fix that, and have all be happy.

gr-sp, 2016-06-07:
I have tried to initialize pybombs prefixes with both relative and absolute paths on CentOS 6.5 and on Ubuntu 14 and neither work for me. If I revert the commit, everything seems to work (both absolute and relative pathing) on both platforms. Out of curiosity, what platform was having issues with this?

gr-sp, 2016-06-07:
FYI I have some guys here working on another project (on Ubuntu 14) and their build failed on a clean install (using relative path to initialize the prefix) when attempting to perform a vanilla gnuradio build.

mbr0wn, 2016-06-07:
Failed for me on Ubuntu 14.04, too. (without the commit).

mbr0wn, 2016-06-07:
Did  	1995909 help?

mbr0wn, 2016-06-07:
also, make sure you have abs paths in all your config.yml files.

gr-sp, 2016-06-07:
Quick initial tests look good...give me about 10 min for more complete check.

gr-sp, 2016-06-07:
Looks good to me. Thanks a bunch!

mbr0wn, 2016-06-07:
cool :)

closed 2016-06-07 by mbr0wn