gnuradio/pybombs #351: Error running pip search

gr-sp, 2016-06-08:
This may be another Python 2 vs Python 3 character encoding issue. I am not really sure because I haven't looked into it and it hasn't really affected me (other than an annoying error output message). The packager is throwing this error when checking for python-requests:

```
PyBombs.Packager.pip - ERROR - Error running pip search
PyBombs.Packager.pip - DEBUG - 'ascii' codec can't encode character u'\u2019' in position 14851: ordinal not in range(128)
```

More context below:
```
PyBombs.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBombs.Recipe[python-requests] - OBNOXIOUS - Loading recipe file: /usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes/python-requests.lwr
PyBombs.Recipe[python-requests] - OBNOXIOUS - Loading template file: /usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/templates/distutils.lwt
PyBombs.Recipe[python-requests] - OBNOXIOUS - Inheriting from file /usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/templates/distutils.lwt
PyBombs.Recipe[python-requests] - OBNOXIOUS - Loading template file: /usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/templates/empty.lwt
PyBombs.Recipe[python-requests] - OBNOXIOUS - Inheriting from file /usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/templates/empty.lwt
PyBombs.Recipe[python-requests] - OBNOXIOUS - Loaded recipe - python-requests
.PyBombs.PackageManager - DEBUG - Checking if package python-requests is installed.
PyBombs.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (python-requests)
PyBombs.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (python-requests)
PyBombs.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (python-requests)
PyBombs.Packager.pip - OBNOXIOUS - Checking if recipe python-requests is installed
PyBombs.ReqScanner - OBNOXIOUS - Adding package with name requests
PyBombs.ReqScanner - OBNOXIOUS - End of requirements list
PyBombs.ReqScanner - OBNOXIOUS - Done parsing requirements string `requests`
PyBombs.Packager.pip - OBNOXIOUS - Calling ev for recursive satisfier rule evaluation
PyBombs.Packager.pip - DEBUG - Loading pip install cache.
PyBombs.Packager.yumdnf - OBNOXIOUS - Checking if recipe python-requests is installed
PyBombs.ReqScanner - OBNOXIOUS - Adding package with name python-requests
PyBombs.ReqScanner - OBNOXIOUS - End of requirements list
PyBombs.ReqScanner - OBNOXIOUS - Done parsing requirements string `python-requests`
PyBombs.Packager.yumdnf - OBNOXIOUS - Calling ev for recursive satisfier rule evaluation
PyBombs.Packager.pkgconfig - OBNOXIOUS - Checking if recipe python-requests is installed
PyBombs.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBombs.Packager.cmd - OBNOXIOUS - Checking if recipe python-requests is installed
PyBombs.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBombs.PackageManager - DEBUG - Checking if package python-requests is installable.
PyBombs.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (python-requests)
PyBombs.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (python-requests)
PyBombs.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (python-requests)
PyBombs.Packager.pip - OBNOXIOUS - exists(python-requests)
PyBombs.ReqScanner - OBNOXIOUS - Adding package with name requests
PyBombs.ReqScanner - OBNOXIOUS - End of requirements list
PyBombs.ReqScanner - OBNOXIOUS - Done parsing requirements string `requests`
PyBombs.Packager.pip - OBNOXIOUS - Calling ev for recursive satisfier rule evaluation
PyBombs.Packager.pip - ERROR - Error running pip search
PyBombs.Packager.pip - DEBUG - 'ascii' codec can't encode character u'\u2019' in position 14851: ordinal not in range(128)
PyBombs.Packager.yumdnf - OBNOXIOUS - exists(python-requests)
PyBombs.ReqScanner - OBNOXIOUS - Adding package with name python-requests
PyBombs.ReqScanner - OBNOXIOUS - End of requirements list
PyBombs.ReqScanner - OBNOXIOUS - Done parsing requirements string `python-requests`
PyBombs.Packager.yumdnf - OBNOXIOUS - Calling ev for recursive satisfier rule evaluation
PyBombs.Packager.yumdnf - DEBUG - Package python-requests has version 2.6.0 in yum
PyBombs.Packager.pkgconfig - OBNOXIOUS - exists(python-requests)
PyBombs.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBombs.Packager.cmd - OBNOXIOUS - exists(python-requests)
```

mbr0wn, 2016-06-08:
Which Python are you running?

mbr0wn, 2016-06-08:
Hm, can't repro right now. I don't get why it's even trying to use the ascii decoding. It's explicitly been told to use utf-8.

gr-sp, 2016-06-08:
I am using 2.7.11 on CentOS 6.5

gr-sp, 2016-06-08:
I built it from source with the following configure step (if this matters):
```
./configure --with-cxx-main=/usr/bin/g++ --enable-shared --enable-ipv6 --enable-unicode=ucs4 --with-ensurepip LDFLAGS="-Wl,--rpath=/usr/local/lib"
```

mbr0wn, 2016-06-09:
Does it make any difference if you leave out the `--enable-unicode=ucs4`? It should default to on, anyway.

gr-sp, 2016-06-09:
I get the error whether or not `--enable-unicode=ucs4` is included in the configure line of the Python build.

mbr0wn, 2016-06-10:
@gr-sp This is the offending code:

```{.py}
 43         try:
 44             out = subproc.check_output(["pip", "search", pkgname])
 45             if len(out) == 0:
 46                 return True
 47             if re.search(r'^\b{pkg}\b'.format(pkg=pkgname), str(out), re.MULTILINE):
 48                 return True
```

I can't reproduce this, though. Can you help me find out if this error is originating in line 44 (that's what I suspect), or line 47? Can you just put in some print statements around those lines and see what happens?

mbr0wn, 2016-06-10:
Actually, we have some code that does that check we should be using. I pushed 100ad884a634ec82818226f72852d6accdc28940 with a small change, somehow, I don't think this'll fix your issue though.

gr-sp, 2016-06-13:
Commit https://github.com/gnuradio/pybombs/commit/100ad884a634ec82818226f72852d6accdc28940 seems to have done the trick. I no longer get the error. Do you still want me to post the output from print statements in the above mentioned code?

mbr0wn, 2016-06-13:
Nah, let's not jinx it. Damned if I know why this worked but not before.

closed 2016-06-13 by mbr0wn