gnuradio/pybombs #352: syntax error: mapping value not allowed

gr-sp, 2016-06-08:
The answer to this may just be "don't do that", but I ran into a weird syntax error the other day when I tried to execute the following recipe:
```
category: baseline
depends:
configure: echo "Test configure: $prefix"
make: echo "Test make"
install: echo "Test install"
uninstall: echo "Test uninstall"
source: /home/<me>/test.tgz
```
(You may need to replace the source line with something valid for you.)

It seems to be choking on the `:` in the configure line. If I remove the colon it stops throwing the error. The above recipes throws the following:
```
Error loading /home/shawn/.pybombs/recipes/ta-recipes/test.lwr: mapping values are not allowed here
  in "<string>", line 3, column 32:
    configure: echo "Test configure: $prefix"
                                   ^
Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 9, in <module>
    load_entry_point('PyBOMBS==2.0.1', 'console_scripts', 'pybombs')()
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/base.py", line 187, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/remove.py", line 88, in run
    if not self.is_installed(pkg):
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/remove.py", line 76, in is_installed
    if self.pm.installed(pkg):
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/package_manager.py", line 153, in installed
    if self.check_package_flag(name, 'forceinstalled'):
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/package_manager.py", line 94, in check_package_flag
    recipe.get_recipe(pkgname).category
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipe.py", line 384, in get_recipe
    r = Recipe(recipe_manager.recipe_manager.get_recipe_filename(pkgname))
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipe.py", line 220, in __init__
    self._data = self._load_recipe_from_file(filename)
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipe.py", line 257, in _load_recipe_from_file
    data = PBConfigFile(filename).get()
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/config_file.py", line 40, in __init__
    assert isinstance(self.data, dict)
AssertionError
```

gr-sp, 2016-06-08:
It would be nice if they had a Q/A section for github projects that allowed for discussion. Anyway, I think this is the last known bug for me. I think a new release would be nice. The only other thing I can think of is that it would be nice if we could use `git describe` to generate the pybombs version number. That way, between releases we would see something like the following:
```
$ pybombs --version
v2.0.1-131-gf23a35a
```

I could put in a new issue if you like.  :)

mbr0wn, 2016-06-08:
A pull request would be even more welcome ;-D

mbr0wn, 2016-06-08:
FYI, you're using YAML wrong. If you want a string value with anything that's funky, you need to escape it with quotes:
```{.yaml}
category: baseline
depends:
configure: 'echo "Test configure: $prefix"'
make: echo "Test make"
install: echo "Test install"
uninstall: echo "Test uninstall"
source: /home/<me>/test.tgz
```

Quoting is like in Python, I believe.

closed 2016-06-08 by mbr0wn