gnuradio/pybombs #357: Subprocess type error

gr-sp, 2016-06-15:
PyBombs.install_manager - INFO - Installing package: autotools-bootstrap
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/utils/subproc.py", line 142, in _process_thread
    args = ' '.join(ELEVATE_PRE_ARGS) + ' ' + args
TypeError: cannot concatenate 'str' and 'list' objects

gr-sp, 2016-06-15:
My build is broken. Can't bootstrap autotools.

gr-sp, 2016-06-15:
Looks like https://github.com/gnuradio/pybombs/commit/c7d90627260dec735cfa616a3d902dab1e900e8c may have done it.

mbr0wn, 2016-06-16:
@noc0lour What was the purpose of c7d9062? I know I'm the one who merged it, but looking at it now I have now idea what this was supposed to fix.

Reverting it for now (dd6de167ac2ed0836ae9f351611cac24364215f7). @gr-sp, that should fix your issue.

gr-sp, 2016-06-16:
Ahhhh...better now.

This worked. I no longer get the above traceback. Thanks!

closed 2016-06-18 by mbr0wn

noc0lour, 2016-06-21:
Sorry for the inconvenience.  Changing it fixed my problem but created new ones. Thanks for fixing it!