gnuradio/pybombs #358: Cannot get a specific revision of recipes.

gr-sp, 2016-06-16:
When performing
```
pybombs recipes add <alias> git+<location/repo.git>
```
the master branch of the requested repository is checked out. However, it would be nice if you could do 
```
pybombs recipes add <alias> git+<location/repo.git> -r <commit|rev|tag>
```
in order to get a specific commit/revision/tag.

mbr0wn, 2016-06-17:
I like this, but I think we can adopt the pip git addressing scheme and append `@<hash>` at the end of the URL. That way, anything fetched by git will get this feature, and we don't add a git-specific switch to the recipes add command.

gr-sp, 2016-06-17:
OK, so that means parsing the URI passed into `fetchers/git.py:fetch_url(...)` if I get your meaning? Sorry, I am not familiar with "pip git addressing scheme".

mbr0wn, 2016-06-17:
Yeah that was a bit ambiguous. What I mean is, instead of

    pybombs recipes add <alias> git+<location/repo.git> -r <commit|rev|tag>

you do

    pybombs recipes add <alias> git+<location/repo.git>@<commit|rev|tag>

Now, the git fetcher handles the URL, including the git rev. For free, you now get more features, e.g. you can specify a source: git...@foo line in a recipe (although admittedly that's redundant because we already have the gitrev option). It also avoids the question of what `-r` means if the source URL is not from git.

mbr0wn, 2016-06-17:
https://coderwall.com/p/-wbo5q/pip-install-a-specific-github-repo-tag-or-branch (FYI re pip addressing)

mbr0wn, 2016-06-18:
e756d7f2a3b3f3c6f433561c77680b7e2bed74d7 provides this in the manner mentioned before. (append `@<hash>`).

closed 2016-06-21 by mbr0wn