gnuradio/pybombs #364: pybombs does not check for sudo

osh, 2016-06-29:
it is assumed to be installed, no check is provided
lack of sudo results in an unhappy execution failed error with no good explanation as to what failed

kmcguire3413, 2016-09-11:
Even with code to not use `sudo` if it does not exist. Does there still need to be a message generated (seems like a good idea) and what type of message is best?

mbr0wn, 2016-09-18:
This is not quite a fix, but I've made the sudo thing a config option. You can now do (in ~/.pybombs/config.yml):
```{.yml}
config:
  elevate_pre_args: [,]
```
and it will not use sudo.

E3V3A, 2017-03-27:
I suggest also using a shell wrapper to pybombs to check for the sudo binary and current `id`.

mbr0wn, 2017-03-28:
shell wrappers have the disadvantage of being non-portable.

mbr0wn, 2017-06-17:
We have a couple of things in place now:
- The `autoconfig` subcommand
- The `elevate_pre_args` config option

I'm considering closing this issue. However, there are still some good ideas:
- A check when *not* to elevate (also, a check *when* to elevate, see #304, but that's a separate issue).
- A better checking of the `elevate_pre_args` value and better error output on failure.