gnuradio/pybombs #365: Issue with setuptools

mhostetter, 2016-07-01:
I'm probably just a n00b, but I always have trouble installing PyBOMBS with setuptools.  I just tried again with a fresh Ubuntu 16.04 machine.  Here is my output.

Am I doing something obviously wrong?

```
matt@series5:~$ git clone https://github.com/gnuradio/pybombs.git
Cloning into 'pybombs'...
remote: Counting objects: 4636, done.
remote: Total 4636 (delta 0), reused 0 (delta 0), pack-reused 4636
Receiving objects: 100% (4636/4636), 1.44 MiB | 2.78 MiB/s, done.
Resolving deltas: 100% (3181/3181), done.
Checking connectivity... done.
matt@series5:~$ cd pybombs/
matt@series5:~/pybombs$ python setup.py build
Downloading https://pypi.python.org/packages/source/s/setuptools/setuptools-23.1.0.zip
Extracting in /tmp/tmpgoMv5E
Traceback (most recent call last):
  File "setup.py", line 71, in <module>
    ez_setup.use_setuptools()
  File "/home/matt/pybombs/ez_setup.py", line 173, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/matt/pybombs/ez_setup.py", line 124, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/matt/pybombs/ez_setup.py", line 64, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/matt/pybombs/ez_setup.py", line 103, in archive_context
    with ContextualZipFile(filename) as archive:
  File "/home/matt/pybombs/ez_setup.py", line 87, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 811, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file
matt@series5:~/pybombs$ 
```

Thanks,
Matt

mbr0wn, 2016-07-05:
@mhostetter What happens if you run it twice? And what happens if you install setuptools through pip by hand first?

mbr0wn, 2016-07-05:
I've removed ez_setup.py -- it's been giving us too much trouble lately, anyway.

mbr0wn, 2016-07-05:
Install setuptools through pip and try latest master. You should be good.

closed 2016-07-05 by mbr0wn

mhostetter, 2016-07-05:
Just for completeness, I ran it a second time and got the same output.

```
matt@series5:~/pybombs$ python setup.py build
Downloading https://pypi.python.org/packages/source/s/setuptools/setuptools-24.0.2.zip
Extracting in /tmp/tmpRejo8y
Traceback (most recent call last):
  File "setup.py", line 71, in <module>
    ez_setup.use_setuptools()
  File "/home/matt/pybombs/ez_setup.py", line 173, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/matt/pybombs/ez_setup.py", line 124, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/matt/pybombs/ez_setup.py", line 64, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/matt/pybombs/ez_setup.py", line 103, in archive_context
    with ContextualZipFile(filename) as archive:
  File "/home/matt/pybombs/ez_setup.py", line 87, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 811, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file
matt@series5:~/pybombs$ 
```

I did install setuptools thought pip and had success.  

Thanks for the help,
Matt