gnuradio/pybombs #366: pybombs failing with 16.04 LTS fresh install

gwratcliff, 2016-07-04:
Greetings,

Martin you asked that I post a -v result here.  I'm using a recently downloaded and installed, with no other changes Ubuntu build from ground up, nothing else.

Added pybombs with pip.
added recipes per the pybombs instructions.
Executed command: pybombs install gnuradio

Results below
gratcliff@gratcliff-Satellite-L650:~$ pybombs -v install gnuradio
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /home/gratcliff/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {'default_prefix': '/home/gratcliff/gnuradio'}
PyBOMBS.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/dist-packages/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/gratcliff/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/home/gratcliff/gnuradio)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/gratcliff/gnuradio/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Config dir does not yet exist.
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix configuration file not found: /home/gratcliff/gnuradio/.pybombs/config.yml, assuming empty.
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/gratcliff/gnuradio/src
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix inventory file not found: /home/gratcliff/gnuradio/.pybombs/inventory.yml
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /home/gratcliff/gnuradio/.pybombs/inventory.yml...
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/home/gratcliff/.pybombs/recipes/gr-etcetera', '/home/gratcliff/.pybombs/recipes/gr-recipes', '/usr/local/lib/python2.7/dist-packages/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBOMBS - INFO - PyBOMBS Version 2.1.0
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/gratcliff/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/gratcliff/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 51 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/gratcliff/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/gratcliff/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 165 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/dist-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/dist-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.install - DEBUG - Initializing command class for command install
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBOMBS.PackageManager - DEBUG - apt-get is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using packagers: ['pip', 'apt-get', 'pkgconfig', 'cmd', 'source']
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBOMBS.PackageManager - DEBUG - apt-get is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using packagers: ['pip', 'apt-get', 'pkgconfig', 'cmd', 'source']
PyBOMBS.PackageManager - DEBUG - Checking if package gnuradio is installed.
PyBOMBS.PackageManager - DEBUG - Package gnuradio is requesting a source build.
[... 346 lines cut ...]
    return dispatch() or 0
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/base.py", line 187, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/commands/install.py", line 122, in run
    static=getattr(self.args, 'static', False),
  File "/usr/local/lib/python2.7/dist-packages/pybombs/install_manager.py", line 85, in install
    if not self.pm.install(pkg, static=static, verify=verify):
  File "/usr/local/lib/python2.7/dist-packages/pybombs/package_manager.py", line 198, in install
    static=static,
  File "/usr/local/lib/python2.7/dist-packages/pybombs/package_manager.py", line 235, in _std_package_operation
    result = getattr(pkgr, operation)(rec, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/packagers/source.py", line 110, in install
    fail_if_builddir_missing=update,
  File "/usr/local/lib/python2.7/dist-packages/pybombs/packagers/source.py", line 240, in run_build
    set_state(self.inventory.STATE_CONFIGURED)
  File "/usr/local/lib/python2.7/dist-packages/pybombs/packagers/source.py", line 208, in <lambda>
    set_state = lambda state: self.inventory.set_state(recipe.id, state) or self.inventory.save()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/inventory.py", line 126, in set_state
    self._invfile.update({pkg: {'state': state}})
  File "/usr/local/lib/python2.7/dist-packages/pybombs/config_file.py", line 57, in update
    self.save()
  File "/usr/local/lib/python2.7/dist-packages/pybombs/config_file.py", line 52, in save
    open(self._filename, 'w').write(yaml.dump(self.data, default_flow_style=False))
IOError: [Errno 2] No such file or directory: '/home/gratcliff/gnuradio/.pybombs/inventory.yml'
gratcliff@gratcliff-Satellite-L650:~$

mbr0wn, 2016-07-05:
@gwratcliff Thanks a lot, very helpful, much appreciated. I think I know what's up -- if you create the directory `/home/gratcliff/gnuradio/.pybombs/` you'll probably be fine. But I'll have a fix up for this soon.

mbr0wn, 2016-07-05:
@gwratcliff Could it be that you're not creating your prefix with `pybombs prefix init`?

mbr0wn, 2016-07-05:
FYI, this line tipped me off:
```
 PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Config dir does not yet exist.
```

mbr0wn, 2016-07-06:
Not sure if this is still an issue....

mbr0wn, 2016-08-31:
Closing -- not sure if still an issue.

closed 2016-08-31 by mbr0wn