gnuradio/pybombs #368: gr-fosphor dependency on gnuradio

mhostetter, 2016-07-09:
Just as an FYI.  I did a fresh install of gnuradio using `$ pybombs install gnuradio` (latest PyBOMBS).  I immediately followed it with `$ pybombs install gr-fosphor`.  The fosphor module apparently depends on gnuradio and rebuilt all of gnuradio from source.  Should gr-fosphor depend on gnuradio?  I know it won't work without it, but...

```
matt@t430:~$ pybombs install gr-fosphor
PyBOMBS - INFO - PyBOMBS Version 2.1.1a
Install tree:
|
\- gr-fosphor
   |
   +- gnuradio
   |  |
   |  +- uhd
   |  |
   |  \- apache-thrift
   |
   \- glfw3
PyBOMBS.install_manager - INFO - Installing package: glfw3
[sudo] password for matt: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-4.4.0-21 linux-headers-4.4.0-21-generic linux-headers-4.4.0-22
  linux-headers-4.4.0-22-generic linux-image-4.4.0-21-generic
  linux-image-4.4.0-22-generic linux-image-extra-4.4.0-21-generic
  linux-image-extra-4.4.0-22-generic
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libglfw3
The following NEW packages will be installed:
  libglfw3 libglfw3-dev
0 upgraded, 2 newly installed, 0 to remove and 197 not upgraded.
Need to get 64.9 kB of archives.
After this operation, 273 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 libglfw3 amd64 3.1.2-3 [38.5 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 libglfw3-dev amd64 3.1.2-3 [26.3 kB]
Fetched 64.9 kB in 0s (220 kB/s)       
Selecting previously unselected package libglfw3:amd64.
(Reading database ... 314905 files and directories currently installed.)
Preparing to unpack .../libglfw3_3.1.2-3_amd64.deb ...
Unpacking libglfw3:amd64 (3.1.2-3) ...
Selecting previously unselected package libglfw3-dev:amd64.
Preparing to unpack .../libglfw3-dev_3.1.2-3_amd64.deb ...
Unpacking libglfw3-dev:amd64 (3.1.2-3) ...
Processing triggers for libc-bin (2.23-0ubuntu3) ...
Setting up libglfw3:amd64 (3.1.2-3) ...
Setting up libglfw3-dev:amd64 (3.1.2-3) ...
Processing triggers for libc-bin (2.23-0ubuntu3) ...
PyBOMBS.install_manager - INFO - Installation successful.
PyBOMBS.install_manager - INFO - Installing package: apache-thrift
Cloning:     (100%) [=========================================================]
Cloning:     (100%) [=========================================================]
Configuring: (100%) [=========================================================]
Building:    (100%) [=========================================================]
Installing:  (100%) [=========================================================]
PyBOMBS.install_manager - INFO - Installation successful.
PyBOMBS.install_manager - INFO - Installing package: uhd
Cloning:     (100%) [=========================================================]
Configuring: (100%) [=========================================================]
Building:    (100%) [=========================================================]]
Installing:  (100%) [=========================================================]
PyBOMBS.install_manager - INFO - Installation successful.
PyBOMBS.install_manager - INFO - Installing package: gnuradio
Cloning:     (100%) [=========================================================]
Configuring: (100%) [=========================================================]
Building:    (100%) [=========================================================]
[  2%] Built target volk_obj
[  2%] Built target volk
[  2%] Built target test_all
[  2%] Built target volk_profile
[  2%] Built target volk-config-info
[  2%] Built target pygen_volk_python_volk_modtool_75df9
[  2%] Built target pygen_volk_python_volk_modtool_2070a
[  2%] Built target pmt_generated
[  3%] Built target gnuradio-pmt
[  7%] Built target gnuradio-runtime
[  7%] Built target test-gnuradio-runtime
[  7%] Built target gr_runtime_test
[  8%] Built target test-gnuradio-pmt
[  8%] Built target gr_pmt_test
[  8%] Built target gnuradio-config-info
[  8%] Built target pmt_swig_swig_doc
[  8%] Built target _pmt_swig_swig_tag
[  8%] Built target pmt_swig_gnuradio_runtime_swig_7dd5e
[  9%] Built target _pmt_swig
[  9%] Built target pmt_swig
[  9%] Built target runtime_swig_swig_doc
Scanning dependencies of target _runtime_swig_swig_tag
[  9%] Building CXX object gnuradio-runtime/swig/CMakeFiles/_runtime_swig_swig_tag.dir/_runtime_swig_swig_tag.cpp.o
[  9%] Linking CXX executable _runtime_swig_swig_tag
[  9%] Built target _runtime_swig_swig_tag
[  9%] Generating runtime_swig.tag
Scanning dependencies of target runtime_swig_gnuradio_runtime_swig_7dd5e
[  9%] Building CXX object gnuradio-runtime/swig/CMakeFiles/runtime_swig_gnuradio_runtime_swig_7dd5e.dir/runtime_swig_gnuradio_runtime_swig_7dd5e.cpp.o
[  9%] Linking CXX executable runtime_swig_gnuradio_runtime_swig_7dd5e
Swig source
[  9%] Built target runtime_swig_gnuradio_runtime_swig_7dd5e
Scanning dependencies of target pygen_gnuradio_runtime_swig_8448e
[  9%] Generating runtime_swig.pyc
[  9%] Generating runtime_swig.pyo
[  9%] Built target pygen_gnuradio_runtime_swig_8448e
Scanning dependencies of target _runtime_swig
[  9%] Building CXX object gnuradio-runtime/swig/CMakeFiles/_runtime_swig.dir/runtime_swigPYTHON_wrap.cxx.o
[  9%] Linking CXX shared module _runtime_swig.so
[  9%] Built target _runtime_swig
[  9%] Built target pygen_gnuradio_runtime_swig_4b7e6
[  9%] Built target pygen_gnuradio_runtime_python_gnuradio_baf0a
[ 10%] Built target pygen_gnuradio_runtime_python_gnuradio_gr_02d8c
[ 10%] Built target pygen_gnuradio_runtime_python_gnuradio_gru_aee9a
[ 10%] Built target pygen_gnuradio_runtime_python_gnuradio_ctrlport_916b8
[ 10%] Built target pygen_gnuradio_runtime_python_gnuradio_ctrlport_e79bc
[ 10%] Built target pygen_gnuradio_runtime_python_gnuradio_ctrlport_1af17
[ 10%] Built target pygen_gnuradio_runtime_python_gnuradio_ctrlport_8e768
[ 10%] Built target pygen_gnuradio_runtime_python_pmt_f9283
[ 10%] Built target pygen_gnuradio_runtime_examples_mp_sched_96e49
[ 11%] Built target pygen_gnuradio_runtime_examples_network_dd92e
[ 11%] Built target pygen_gnuradio_runtime_examples_volk_benchmark_ea9ab
[ 13%] Built target blocks_generated_includes
[ 30%] Built target gnuradio-blocks
Scanning dependencies of target test-gr-blocks
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/test_gr_blocks.cc.o
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/qa_gr_flowgraph.cc.o
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/qa_set_msg_handler.cc.o
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/qa_gr_block.cc.o
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/qa_gr_top_block.cc.o
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/qa_gr_hier_block2.cc.o
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/qa_gr_hier_block2_derived.cc.o
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/qa_blocks.cc.o
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/qa_block_tags.cc.o
[ 31%] Building CXX object gr-blocks/lib/CMakeFiles/test-gr-blocks.dir/qa_rotator.cc.o
[ 31%] Linking CXX executable test-gr-blocks
[ 31%] Built target test-gr-blocks
[ 31%] Built target pygen_gr_blocks_python_blocks_09f7e
[ 31%] Built target pygen_gr_blocks_python_grc_gnuradio_eac7f
[ 31%] Built target pygen_gr_blocks_python_grc_gnuradio_2377d
[ 31%] Built target pygen_gr_blocks_swig_11144
[ 31%] Built target blocks_swig5_swig_doc
[ 31%] Built target _blocks_swig5_swig_tag
[ 31%] Built target blocks_swig5_gr_blocks_swig_a6e57
[ 32%] Built target pygen_gr_blocks_swig_66378
Scanning dependencies of target _blocks_swig5
[ 32%] Building CXX object gr-blocks/swig/CMakeFiles/_blocks_swig5.dir/blocks_swig5PYTHON_wrap.cxx.o
[ 32%] Linking CXX shared module _blocks_swig5.so
[ 32%] Built target _blocks_swig5
[ 32%] Built target blocks_swig4_swig_doc
[ 32%] Built target _blocks_swig4_swig_tag
[ 33%] Built target blocks_swig4_gr_blocks_swig_a6e57
...
```

mhostetter, 2016-07-09:
Just for completeness, this was my original install of gnuradio.

```
matt@t430:~/pybombs$ pybombs install gnuradio
PyBOMBS - INFO - PyBOMBS Version 2.1.1a
Install tree:
|
\- gnuradio
   |
   +- uhd
   |
   \- apache-thrift
PyBOMBS.install_manager - INFO - Installing package: apache-thrift
Cloning:     (100%) [=========================================================]
Cloning:     (100%) [=========================================================]
Configuring: (100%) [=========================================================]
Building:    (100%) [=========================================================]
Installing:  (100%) [=========================================================]
PyBOMBS.install_manager - INFO - Installation successful.
PyBOMBS.install_manager - INFO - Installing package: uhd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-4.4.0-21 linux-headers-4.4.0-21-generic linux-headers-4.4.0-22
  linux-headers-4.4.0-22-generic linux-image-4.4.0-21-generic
  linux-image-4.4.0-22-generic linux-image-extra-4.4.0-21-generic
  linux-image-extra-4.4.0-22-generic
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libgps22 libuhd003
Suggested packages:
  gpsd gnuradio
The following NEW packages will be installed:
  libgps22 libuhd003 uhd-host
0 upgraded, 3 newly installed, 0 to remove and 197 not upgraded.
Need to get 3,730 kB of archives.
After this operation, 26.2 MB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 libgps22 amd64 3.15-2build1 [51.2 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 libuhd003 amd64 3.9.2-1 [1,755 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 uhd-host amd64 3.9.2-1 [1,924 kB]
Fetched 3,730 kB in 1s (2,624 kB/s)   
Selecting previously unselected package libgps22:amd64.
(Reading database ... 308698 files and directories currently installed.)
Preparing to unpack .../libgps22_3.15-2build1_amd64.deb ...
Unpacking libgps22:amd64 (3.15-2build1) ...
Selecting previously unselected package libuhd003:amd64.
Preparing to unpack .../libuhd003_3.9.2-1_amd64.deb ...
Unpacking libuhd003:amd64 (3.9.2-1) ...
Selecting previously unselected package uhd-host.
Preparing to unpack .../uhd-host_3.9.2-1_amd64.deb ...
Unpacking uhd-host (3.9.2-1) ...
Processing triggers for libc-bin (2.23-0ubuntu3) ...
Processing triggers for man-db (2.7.5-1) ...
Processing triggers for doc-base (0.10.7) ...
Processing 1 added doc-base file...
Setting up libgps22:amd64 (3.15-2build1) ...
Setting up libuhd003:amd64 (3.9.2-1) ...
Setting up uhd-host (3.9.2-1) ...
Adding group `usrp' (GID 129) ...
Done.
Processing triggers for libc-bin (2.23-0ubuntu3) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-4.4.0-21 linux-headers-4.4.0-21-generic linux-headers-4.4.0-22
  linux-headers-4.4.0-22-generic linux-image-4.4.0-21-generic
  linux-image-4.4.0-22-generic linux-image-extra-4.4.0-21-generic
  linux-image-extra-4.4.0-22-generic
[... 237 lines cut ...]
Install icon: 32x32
Install icon: 48x48
Install icon: 64x64
Install icon: 128x128
Install icon: 256x256
Install mime type
Install menu items
Done!
Setting up libvolk1-dev (1.2.1-1) ...
Setting up gnuradio-dev (3.7.9.1-2ubuntu1) ...
Setting up libuhd-dev (3.9.2-1) ...
Setting up python-networkx (1.11-1ubuntu1) ...
Setting up python-yaml (3.11-3build1) ...
Setting up librtlsdr0:amd64 (0.5.3-5) ...
Setting up rtl-sdr (0.5.3-5) ...
Processing triggers for libc-bin (2.23-0ubuntu3) ...
PyBOMBS.install_manager - INFO - Installation successful.
PyBOMBS.install_manager - INFO - Installing package: gnuradio
Cloning:     (100%) [=========================================================]
Configuring: (100%) [=========================================================]
Building:    (100%) [=========================================================]]
Installing:  (100%) [=========================================================]
PyBOMBS.install_manager - INFO - Installation successful.
matt@t430:~/pybombs$ 
```

mbr0wn, 2016-07-11:
Hm, not sure what's going on here. PyBOMBS should not rebuild GNU Radio like that.

mbr0wn, 2016-12-01:
@mhostetter A lot has changed since you posted this. I tried reproducing on the current HEAD, and it works fine, which may be something I accidentally fixed.
Closing for now -- if you still have this issue, please ping this thread and I'll reopen.

closed 2016-12-01 by mbr0wn

mhostetter, 2016-12-01:
@mbr0wn thanks for coming back to this.  I've been away for a while, but I just pulled the latest PyBOMBS.  Looking really good!  Thanks for all the hard work on this.

mbr0wn, 2016-12-01:
You're very welcome!