gnuradio/pybombs #369: Use pkexec instead of sudo for granting elevated privileges

NinjaComics, 2016-07-12:
It would be nice if we use pkexec instead of sudo (we currently use sudo -H in subproc.py) for granting the elevated privileges while installing packages using pybombs. On the upside, pkexec will call the default system authentication agent and prompts the user to enter the password in a gui dialog if the app is running in X session or prompts to enter in terminal if the app is run from a tty. Plus, this will work for both pybombs-cli and the pybombs-gui.

On the downside, everytime the pkexec is called from within pybombs, the user has to enter the password. 

Example:
**sudo:** 

`sudo <do-something> #asks for password`
`sudo <do-something-after-a-little-while> #executes the command without asking for password`

**pkexec:**

`pkexec <do-something> #asks for password`
`pkexec <do-something-after-a-little-while> #asks for password`

pkexec does that for security reasons.

This problem can be overcome by writing policy files as specified in [pkexec's reference manual](https://www.freedesktop.org/software/polkit/docs/0.105/pkexec.1.html#pkexec-example) for each packager that requires elevated privileges.

mbr0wn, 2016-12-12:
With `elevate_pre_args`, we can already do this. Would make sense to define policy files though if we really want to support this, or it's kinda pointless. Do you want to work on this?

NinjaComics, 2016-12-12:
Enabling policies for each packager makes sense only when pybombs is in use. When not in use, I would want to restore the normal functionality of the packager. I will work on it and submit a PR soon.

mbr0wn, 2017-07-09:
@NinjaComics Did you ever submit that PR?