gnuradio/pybombs #372: Centos 7 error with wxpython install

sgardner340, 2016-07-19:
11:34:22 make: *** [coredll_imagpng.o] Error 1
11:34:24 PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
11:34:24 PyBOMBS.Packager.source - ERROR - Problem occurred while building package wxpython:
11:34:24 Build failed.
11:34:24 PyBOMBS.install_manager - ERROR - Error installing package wxpython. Aborting.

sgardner340, 2016-07-19:
Full Error:
11:34:22 ./src/common/imagpng.cpp: In member function ‘virtual bool wxPNGHandler::LoadFile(wxImage*, wxInputStream&, bool, int)’:
11:34:22 ./src/common/imagpng.cpp:532:30: error: ‘voidp’ was not declared in this scope
11:34:22                              (voidp) NULL,
11:34:22                               ^
11:34:22 ./src/common/imagpng.cpp:589:40: error: invalid use of incomplete type ‘png_info {aka struct png_info_def}’
11:34:22          const size_t ncolors = info_ptr->num_palette;
11:34:22                                         ^
11:34:22 In file included from ./src/common/imagpng.cpp:36:0:
11:34:22 /usr/include/libpng15/png.h:730:16: error: forward declaration of ‘png_info {aka struct png_info_def}’
11:34:22  typedef struct png_info_def png_info;
11:34:22                 ^
11:34:22 ./src/common/imagpng.cpp:596:28: error: invalid use of incomplete type ‘png_info {aka struct png_info_def}’
11:34:22              r[j] = info_ptr->palette[j].red;
11:34:22                             ^
11:34:22 In file included from ./src/common/imagpng.cpp:36:0:
11:34:22 /usr/include/libpng15/png.h:730:16: error: forward declaration of ‘png_info {aka struct png_info_def}’
11:34:22  typedef struct png_info_def png_info;
11:34:22                 ^
11:34:22 ./src/common/imagpng.cpp:597:28: error: invalid use of incomplete type ‘png_info {aka struct png_info_def}’
11:34:22              g[j] = info_ptr->palette[j].green;
11:34:22                             ^
11:34:22 In file included from ./src/common/imagpng.cpp:36:0:
11:34:22 /usr/include/libpng15/png.h:730:16: error: forward declaration of ‘png_info {aka struct png_info_def}’
11:34:22  typedef struct png_info_def png_info;
11:34:22                 ^
11:34:22 ./src/common/imagpng.cpp:598:28: error: invalid use of incomplete type ‘png_info {aka struct png_info_def}’
11:34:22              b[j] = info_ptr->palette[j].blue;
11:34:22                             ^
11:34:22 In file included from ./src/common/imagpng.cpp:36:0:
11:34:22 /usr/include/libpng15/png.h:730:16: error: forward declaration of ‘png_info {aka struct png_info_def}’
11:34:22  typedef struct png_info_def png_info;
11:34:22                 ^
11:34:22 make: *** [coredll_imagpng.o] Error 1
11:34:24 PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
11:34:24 PyBOMBS.Packager.source - ERROR - Problem occurred while building package wxpython:
11:34:24 Build failed.
11:34:24 PyBOMBS.install_manager - ERROR - Error installing package wxpython. Aborting.

mbr0wn, 2016-11-18:
Seems to install fine from binary packages. Closing for now.

closed 2016-11-18 by mbr0wn