gnuradio/pybombs #376: Installing GNURadio failed on Ubuntu 16.04

bd5rv, 2016-07-26:
Installed a fresh pybombs and then
pybombs install gnuradio

It stopped as the followings:

[ 85%] Building CXX object gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o
/opt/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/opt/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:27884:15: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (arg1)->set_gpio_debug(arg2,arg3);
               ^
/opt/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_sptr_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/opt/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:29250:16: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (*arg1)->set_gpio_debug(arg2,arg3);
                ^
/opt/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘void init_uhd_swig()’:
/opt/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:48435:91: error: ‘ATR_REG_IDLE’ is not a member of ‘uhd::usrp::dboard_iface’
 ant(d, "dboard_iface_ATR_REG_IDLE",SWIG_From_int(static_cast< int >(uhd::usrp::
                                                                     ^
/opt/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:48436:94: error: ‘ATR_REG_TX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
 (d, "dboard_iface_ATR_REG_TX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::
                                                                     ^
/opt/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:48437:94: error: ‘ATR_REG_RX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
 (d, "dboard_iface_ATR_REG_RX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::
                                                                     ^
/opt/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:48438:98: error: ‘ATR_REG_FULL_DUPLEX’ is not a member of ‘uhd::usrp::dboard_iface’
 "dboard_iface_ATR_REG_FULL_DUPLEX",SWIG_From_int(static_cast< int >(uhd::usrp::
                                                                     ^
gr-uhd/swig/CMakeFiles/_uhd_swig.dir/build.make:68: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o' failed
make[2]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:12072: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all' failed
make[1]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.

mbr0wn, 2016-07-29:
This looks like there's another GNU Radio or UHD installation somewhere.

u0m3, 2016-08-02:
So this is not an intended use? (having an OS install and one or more pybombs builds)

PeterTNguyen, 2016-08-10:
I had this issue today, but it turns out I had libuhd-dev installed via apt-get. Promptly removed and the issue was resolved.

mbr0wn, 2016-08-31:
Yeah I thought that would be the case. So, this is a bug in GNU Radio (maybe gr-uhd, maybe general SWIG). We can't solve it from here.

closed 2016-08-31 by mbr0wn

codenoid, 2017-10-30:
@mbr0wn , not fixed with @PeterTNguyen comment, 

```
[ 87%] Building CXX object gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o
/home/susu/workarea-gnuradio/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/home/susu/workarea-gnuradio/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:27884:15: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (arg1)->set_gpio_debug(arg2,arg3);
               ^
/home/susu/workarea-gnuradio/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_sptr_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/home/susu/workarea-gnuradio/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:29250:16: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (*arg1)->set_gpio_debug(arg2,arg3);
                ^
/home/susu/workarea-gnuradio/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘void init_uhd_swig()’:
/home/susu/workarea-gnuradio/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:49465:91: error: ‘ATR_REG_IDLE’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_IDLE",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_IDLE)));
                                                                                           ^
/home/susu/workarea-gnuradio/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:49466:94: error: ‘ATR_REG_TX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_TX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_TX_ONLY)));
                                                                                              ^
/home/susu/workarea-gnuradio/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:49467:94: error: ‘ATR_REG_RX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_RX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_RX_ONLY)));
                                                                                              ^
/home/susu/workarea-gnuradio/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:49468:98: error: ‘ATR_REG_FULL_DUPLEX’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_FULL_DUPLEX",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_FULL_DUPLEX)));
                                                                                                  ^
gr-uhd/swig/CMakeFiles/_uhd_swig.dir/build.make:68: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o' failed
make[2]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:13088: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all' failed
make[1]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

codenoid, 2017-10-30:
```
susu@kambing:~/workarea-gnuradio/gnuradio/build$ cat /etc/*release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.3 LTS"
NAME="Ubuntu"
VERSION="16.04.3 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.3 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
```

eriklacson, 2017-12-14:
Getting exactly the same error.

erik@archangel:~/sdr$ cat /etc/*release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.1 LTS"
NAME="Ubuntu"
VERSION="16.04.1 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.1 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial