gnuradio/pybombs #377: PyBOMBS install fails on some clean docker images

stwunsch, 2016-07-27:
Hi!

I've done a fork of pybombs and added some Dockerfiles to test pybombs on different linux distributions ([see here](https://github.com/stwunsch/pybombs/tree/docker)). These files are automatically loaded by hub.docker.com and used to build a new image of the linux dist + pybombs ([see here](https://hub.docker.com/r/stwunsch/pybombs/)). This is done for Ubuntu 12.04, 14.04, 16.04, Fedora 23, 24 and CentOS 6, 7 using the standard pip installation and a installation using the git master version of pybombs.

**Note:** Again, what the Dockerfiles do: They setup the minimum of required packages (python-pip for pip installation and python-pip and git for git installation), init a prefix and then pull the recipes. And that's all.
If this runs smoothly, you can use the images to test a full gnuradio installation on each linux distribution. Unfortunately, this takes too long so that the automated build on hub.docker.com stops after 2h.

So, there are some issues (error log at the bottom of each page, look at the Dockerfiles for the configuration):

**CentOS 7/pip**: [see here](https://hub.docker.com/r/stwunsch/pybombs/builds/bpezxhprmfjdqwjfhxuhrgs/)
**CentOS 6/pip**: [see here](https://hub.docker.com/r/stwunsch/pybombs/builds/bh9ovh3qzqz7z6uvb4i2mcd/)
**CentOS 6/git**: [see here](https://hub.docker.com/r/stwunsch/pybombs/builds/bnwxaghudawmxpvwhigutzw/)
**Ubuntu 12.04/pip**: [see here](https://hub.docker.com/r/stwunsch/pybombs/builds/bdk6zt5gw3vzrf42mppulsh/)
**Ubuntu 16.04/pip**: [see here](https://hub.docker.com/r/stwunsch/pybombs/builds/b4a86jc95gr3rrya4hgdhp3/)

Probably you want to add something like this to the official pybombs repo? This would prevent problems with some frequently used linux flavours before someone complains about problems (note that Ubuntu 14.04 works fine). hub.docker.com would load the Dockerfiles automatically after a new commit on this repo and creates the images.

Greetings!

mbr0wn, 2016-07-29:
Can you run 5079ab4a8aebe257c77e095b7b05e9f5339486e0 real quick? Has no fix, but it has better error message handling. Your error is super weird -- it makes no sense. I'm wondering if maybe something is mangling the command line arguments.

stwunsch, 2016-07-29:
Yep no problem, I rebased my fork. [Here](https://hub.docker.com/r/stwunsch/pybombs/builds/) you can check the new images.

mbr0wn, 2016-08-31:
@stwunsch Sorry for letting this hang around for so long. Actually, when I first looked at this I didn't realize how awesome your setup is.

CentOS 6 uses Python 2.6 which is not supported. I wouldn't mind if someone adds 2.6 support if it's not a pain (and doesn't screw up the code), but I won't do it and PB is officially flagged as requiring Python 2.7 minimum. Although... we could install Python from source. But let's consider Python 2.6 unsupported.

As for the other issues, I think I fixed them, but you're telling pip to install the latest release, which doesn't contain those fixes. I should tag another release... but I'll probably wait for the GRCon hackfest.

stwunsch, 2016-08-31:
Thx. Just wanted to point out the possibilities with docker and simple dockerfiles.

Actually, this could get really interesting in combination with jenkin pipelines and github triggers. E.g., a gnuradio commit triggers a build on top of the pybombs docker image(s) and pybombs commits trigger the build of images from docker base images of the different distros, and so on. If nothing breaks there, it won't break most likely on the productive system. I think Jonathan told me that he uses already sth similar but not integrated in a CI framework. You can even think of adding a publishing step to the pipeline if no error occured and push it to docker hub.

Feel free to close this issue!

jmcorgan, 2016-08-31:
If you're referring to me (Johnathan, not Jonathan :), yes indeed I do use docker extensively to test GNU Radio builds in different OS/configurations.  While I don't use PyBOMBS for this, I'd be very happy for anyone to take the initiative to get automated builds working with something like Travis CI and their Docker integration, tied into our github pull request system.

I have some reservations, however, about what we eventually publish as Dockerfiles on Docker Hub, as there are many use cases that call for different configurations, and it is easy to mistake one particular thing that "works for me" as applying to the entire broader base of users we may have.

stwunsch, 2016-08-31:
Next time I just use a tag for the name :P It's true that a Travis integration for the pull request would be awesome! Some orgs have even a builder which talks to them in the commits and can be asked to do specific checks, e.g., look [here](https://github.com/tensorflow/tensorflow/pull/3865). Future is here!

mbr0wn, 2016-08-31:
OK, looks like the initial issue is no more and we can close this. Still, yanking PRs into automated tests would be fantastic.

closed 2016-08-31 by mbr0wn

stwunsch, 2016-08-31:
If I find some time, probably I give Travis CI a chance!

jmcorgan, 2016-08-31:
Thanks.  Yeah, the bitcoin project has been using travis ci/github integration for a long time, just not the docker part.  Anyway, we should take this conversation out of this PR to elsewhere.