gnuradio/pybombs #379: ENABLE_RFNOC set to OFF

PeterTNguyen, 2016-08-10:
I got this issue doing a fresh install of gnuradio. It looks like the rfnoc folder from uhd doesn't get made/copied to the include folder because ENABLE_RFNOC is set to OFF. I ended up just copying the folder to include, which worked for me.

[ 84%] Building CXX object gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_block_impl.cc.o
In file included from /home/peter/openlte/include/uhd/usrp/multi_usrp.hpp:38:0,
                 from /home/peter/openlte/src/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_block.h:28,
                 from /home/peter/openlte/src/gnuradio/gr-uhd/lib/usrp_block_impl.h:26,
                 from /home/peter/openlte/src/gnuradio/gr-uhd/lib/usrp_block_impl.cc:23:
/home/peter/openlte/include/uhd/device3.hpp:22:31: fatal error: uhd/rfnoc/graph.hpp: No such file or directory
 #include <uhd/rfnoc/graph.hpp>
                               ^
compilation terminated.
make[2]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_block_impl.cc.o] Error 1
make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.

mbr0wn, 2016-08-10:
@PeterTNguyen It's not clear what you did to end up with this error. This line:
```
 /home/peter/openlte/include/uhd/device3.hpp:22:31: fatal error: uhd/rfnoc/graph.hpp: No such file or directory
```
implies that you're actually trying to build gr-ettus, but that wouldn't make sense with master branch. On rfnoc-devel branch, this would work. What was your initial command to kick this off?

mbr0wn, 2016-08-10:
I fixed a bug on UHD master that would falsely enable UHD_RFNOC_FOUND, but that won't fix your setup, only give you a nicer error message.

closed 2016-08-10 by mbr0wn

mbr0wn, 2016-08-11:
@PeterTNguyen I misread your error message -- this is a UHD error and I have a fix for it.