gnuradio/pybombs #383: PyBOMBS.Packager.source - ERROR - Problem occurred while building package rtl-sdr

OliverFaust, 2016-09-13:
Dear Experts,

I have an issue with the PyBOMBS installation procedure as published in the readme file on GitHub.
Here is the relevant console output:
oliver@oliver-VirtualBox:~$ **sudo pybombs prefix init /usr/local/t -a myprefix -R gnuradio-default**
PyBOMBS - INFO - PyBOMBS Version 2.1.0
PyBOMBS.prefix - INFO - Creating directory `/usr/localt'
Alias `myprefix' already exists, overwrite Y/[N]? y
PyBOMBS.ConfigManager - INFO - Creating new config file /usr/localt/.pybombs/config.yml
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
  - gr-osmosdr
The directory '/home/oliver/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Install tree:
PyBOMBS.install_manager - INFO - Installing package: rtl-sdr
Cloning:     (100%) [=========================================================]
**PyBOMBS.Packager.source - ERROR - Problem occurred while building package rtl-sdr:
Unable to fetch recipe rtl-sdr
PyBOMBS.install_manager - ERROR - Error installing package rtl-sdr. Aborting.**

Any help is greatly appreciated.

Oliver

mbr0wn, 2016-09-13:
@OliverFaust The problem is that the download is failing. Maybe it's a firewall issue on your end, or maybe the rtlsdr source is actually unavailable temporarily.

Two observations: Running all of PyBOMBS with `sudo` is not recommended, although right now that's still the only way to get stuff into /usr/local. Also, it looks like you have a typo in your path `/usr/localt`.

closed 2016-09-13 by mbr0wn

OliverFaust, 2016-09-13:
Dear mbr0wn,

Thank you for the prompt response.
Is there a workaround for that problem? 
I am not particularly interested in rtl-sdr, I am interested in installing gnuradio via PIBOMBS. So, it might help to switch off fetching and installing that functionality.

Oliver

mbr0wn, 2016-09-13:
Run with `-v` and see what exactly the problem is.

mbr0wn, 2016-09-13:
Do a `pybombs recipes update` and try again. I've removed gr-osmosdr as a dependency.

OliverFaust, 2016-09-13:
Dear mbr0wn,

Here is the verbose output:
PyBOMBS.Fetcher.git - DEBUG - Using url - git://git.osmocom.org/rtl-sdr
PyBOMBS._process_thread() - DEBUG - Executing command `['git', 'clone', 'git://git.osmocom.org/rtl-sdr', 'rtl-sdr', '-b', 'master']'
Cloning:     (100%) [=========================================================]
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 128
PyBOMBS.Fetcher - DEBUG - That didn't work.
PyBOMBS.Fetcher - DEBUG - Process returned value: 128
PyBOMBS.Packager.source - ERROR - Problem occurred while building package rtl-sdr:
Unable to fetch recipe rtl-sdr
PyBOMBS.install_manager - ERROR - Error installing package rtl-sdr. Aborting

Oliver

OliverFaust, 2016-09-15:
Dear mbr0wn,

It took a new WB image, but GNURADIO is now correctly installed. Thank you very much.

Oliver