gnuradio/pybombs #385: CentOS 7 (in docker): PyBOMBS.recipes - ERROR - Illegal recipes command: add

Flamewires, 2016-09-16:
<pre>
user@host$ sudo docker run --net=host -it centos /bin/bash
[root@host /]# yum install epel-release

Loaded plugins: fastestmirror, ovl
< ... snip ... >
Installed:
  epel-release.noarch 0:7-6                                                     

Complete!
[root@host /]# yum install python-pip
Loaded plugins: fastestmirror, ovl
< ... snip ... >
Installed:
  python-pip.noarch 0:7.1.0-1.el7                                               

Dependency Installed:
  python-backports.x86_64 0:1.0-8.el7                                           
  python-backports-ssl_match_hostname.noarch 0:3.4.0.2-4.el7                    
  python-setuptools.noarch 0:0.9.8-4.el7                                        

Complete!
[root@host /]# pip install pybombs
You are using pip version 7.1.0, however version 8.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Collecting pybombs
  Downloading PyBOMBS-2.1.0.tar.gz (63kB)
    100% |################################| 65kB 703kB/s 
Requirement already satisfied (use --upgrade to upgrade): setuptools in /usr/lib/python2.7/site-packages (from pybombs)
Collecting PyYAML (from pybombs)
  Downloading PyYAML-3.12.tar.gz (253kB)
    100% |################################| 253kB 228kB/s 
Collecting requests (from pybombs)
  Downloading requests-2.11.1-py2.py3-none-any.whl (514kB)
    100% |################################| 516kB 446kB/s 
Collecting six (from pybombs)
  Downloading six-1.10.0-py2.py3-none-any.whl
Collecting future (from pybombs)
  Downloading future-0.15.2.tar.gz (1.6MB)
    100% |################################| 1.6MB 158kB/s 
Installing collected packages: PyYAML, requests, six, future, pybombs
  Running setup.py install for PyYAML
  Running setup.py install for future
  Running setup.py install for pybombs
Successfully installed PyYAML-3.12 future-0.15.2 pybombs-2.1.0 requests-2.11.1 six-1.10.0
[root@host /]# pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git 
PyBOMBS - INFO - PyBOMBS Version 2.1.0
PyBOMBS.recipes - ERROR - Illegal recipes command: add
[root@host /]# 
</pre>

Thoughts?

Flamewires, 2016-09-17:
So did a little debugging. This is due to git not being present. <code>yum install git</code> before attempting to add recipes fixes the problem.

Any way to add a check for git earlier in the chain so the user at least gets a more appropriate warning? This would only need to be done if the recipes are attempted to be fetched using git (git+http(s)) . 

Note: This was using the pip version of PyBOMBS, so if there is a check already, but hasn't made it into pip, then disregard. 

After this, there was more missing stuff, but once we have recipes, I believe the rest of the deps can be handled through additional deps. I'll open another ticket for those.

mbr0wn, 2016-09-18:
Despite what  I said at first, this was an awesome bug report, thanks! I think the git version of PyBOMBS (which you can also install through pip) has better error reporting, but if you had used that, you would not have found this issue :)

mbr0wn, 2016-10-16:
Not sure how this was resolved, but it doesn't seem to happen anymore.

closed 2016-10-16 by mbr0wn