gnuradio/pybombs #386: gr-pocsag won't install.

dohzer, 2016-09-17:
When attempting to install the gr-pocsag package, I get a" connection refused message:

> dohzer@dohzerUbuntu:~$ pybombs install gr-pocsag
PyBOMBS - INFO - PyBOMBS Version 2.1.0
Install tree:
|
\- gr-pocsag
PyBOMBS.install_manager - INFO - Installing package: gr-pocsag
svn: E170013: Unable to connect to a repository at URL 'https://www.cgran.org/svn/projects/gr-pocsag/trunk'
svn: E000111: Error running context: Connection refused
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-pocsag:
There should be a source dir in /home/dohzer/dev/pybombs/src/gr-pocsag, but there isn't.
PyBOMBS.install_manager - ERROR - Error installing package gr-pocsag. Aborting.

From memory, my last attempt at installing gr-pocsag using PyBombs approximatey a year ago had the same problem. Loading the link in the error message with a browser produces a similar "Connection Refused" message:

> This site can’t be reached
www.cgran.org refused to connect.
Try:
Checking the connection
Checking the proxy and the firewall
ERR_CONNECTION_REFUSED

ckuethe, 2016-09-17:
Once upon a time CGRAN hosted repositories, but this is no longer the case. I do have a backup of all the old CGRAN repositories, but it'll take a while to extract gr-pocsag and upload it here.

mbr0wn, 2016-09-18:
I think I have the correct gr-pocsag URL in the recipe now. If it's still wrong, please post an issue on gr-etcetera.

closed 2016-09-18 by mbr0wn