gnuradio/pybombs #396: Ubuntu 14.04: Strange errors resolved with repeated 'rm -rf' and pybombs cycling

ttsou, 2016-10-03:
Reported by Andrew Lanezn on GNU Radio mailing list when installing RFNoC with pybombs. Reproduced by @ttsou.

Andrew Lanez wrote:
>I finally got everything installed successfully after doing 30+ attempts (this is no exaggeration) of rm -rf rfnoc and pybombs prefix init ~/rfnoc -R rfnoc -a alias. I was getting wildly varying errors from packages not being found (though I could manually hit those websites and download the files) to compiling errors with make scripts.

https://lists.gnu.org/archive/html/discuss-gnuradio/2016-09/msg00158.html

**Example error**
```
make[2]: Entering directory `/home/switchlanez/rfnoc/src/pyqt4/qpy/QtCore'
g++ -c -m64 -pipe -fno-strict-aliasing -O2 -fPIC -Wall -W -D_REENTRANT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I/usr/include/python2.7 -I../../QtCore -I. -I. -o qpycore_chimera.o qpycore_chimera.cpp
qpycore_chimera.cpp: In member function ‘void Chimera::set_flag()’:
qpycore_chimera.cpp:376:50: error: ‘pyqt4ClassTypeDef’ has no member named ‘qt4_flags’
         _is_flag = ((pyqt4ClassTypeDef *)_type)->qt4_flags & 0x01;
                                                  ^
make[2]: *** [qpycore_chimera.o] Error 1
make[2]: Leaving directory `/home/switchlanez/rfnoc/src/pyqt4/qpy/QtCore'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/switchlanez/rfnoc/src/pyqt4/qpy'
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package pyqt4:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package pyqt4. Aborting.
```

mbr0wn, 2016-10-08:
In the example error, you're building pyqt4, which shouldn't happen on an Ubuntu 14.04. Hmm....

mbr0wn, 2016-10-08:
Also, I don't think this happens when doing `-R gnuradio-default`. Hrm??

courtarro, 2016-10-12:
This is a bug in the stuff I contributed a few weeks ago. Python-apt is not updated after an install, so it has cached the status of newly-installed packages as being uninstalled. I will need to fix this by updating the cache anytime we install a package.

PR #399 (2016-10-13): Update the python-apt cache whenever a package is installed.
Fixes #396 and possibly others.
commit a00e36e8 (2016-10-13): Update the python-apt cache whenever a package is installed. Fixes #396 and possibly others.
closed 2016-10-13 by mbr0wn