gnuradio/pybombs #401: Wrong execution of apt

ly2ss, 2016-10-15:
pybombs version 2.2.0, mint18 (ubuntu xenial)
exerpt:

PyBOMBS.PackageManager - DEBUG - Checking if package qt4 is installed.
PyBOMBS.install_manager - INFO - Installing package: qt4
PyBOMBS.PackageManager - DEBUG - install(qt4, static=False)
PyBOMBS.PackageManager - DEBUG - Using packager pip
PyBOMBS.PackageManager - DEBUG - Using packager apt
PyBOMBS.Packager.apt - DEBUG - Package libqt4-dev has version 4.8.7 in repositories
PyBOMBS.monitor_process() - DEBUG - Running with elevated privileges.
PyBOMBS._process_thread() - DEBUG - Executing command `['sudo', '-H', 'apt', '-y', 'install', 'libqt4-dev']'
apt
Usage: apt command [options]
       apt help command [options]

Commands:
  add-repository   - Add entries to apt sources.list
  autoclean        - Erase old downloaded archive files
  autoremove       - Remove automatically all unused packages
  build            - Build binary or source packages from sources
  build-dep        - Configure build-dependencies for source packages
  changelog        - View a package's changelog
.
.
.
yBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.PackageManager - ERROR - Something went wrong while trying to install qt4 using apt: Process returned value: 1
PyBOMBS.PackageManager - DEBUG - Using packager pkgconfig
PyBOMBS.PackageManager - DEBUG - Using packager cmd
PyBOMBS.PackageManager - DEBUG - Using packager source
PyBOMBS.Packager.source - DEBUG - State on package qt4 is 0
PyBOMBS.Fetcher - DEBUG - Fetching source for recipe: qt4
PyBOMBS.Fetcher - DEBUG - Switching to src directory: /home/zilvis/prefix/src
PyBOMBS.Fetcher - DEBUG - Getting fetcher for https://download.qt.io/archive/qt/4.6/qt-everywhere-opensource-src-4.6.2.tar.gz, type wget
PyBOMBS.Fetcher - DEBUG - Using fetcher <pybombs.fetchers.wget.Wget object at 0x7fc40386f9e8>
PyBOMBS.Fetcher.wget - DEBUG - Requiring packages on host system: ['python-requests']
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.PackageManager - DEBUG - apt is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using packagers: ['pip', 'apt', 'pkgconfig', 'cmd']
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.PackageManager - DEBUG - apt is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using packagers: ['pip', 'apt', 'pkgconfig', 'cmd']
PyBOMBS.Fetcher.wget - DEBUG - Requirements met.
PyBOMBS.Fetcher.wget - DEBUG - Downloading file: https://download.qt.io/archive/qt/4.6/qt-everywhere-opensource-src-4.6.2.tar.gz

RemiSlv, 2016-10-15:
I'm having the same issue with Linux Mint 18

mbr0wn, 2016-10-15:
I'm considering backing out raw apt support. It doesn't seem to help anything.

mbt25, 2016-10-16:
when it calls apt, it calls it with a "-y", which causes it to fail.   I edited /usr/local/lib/python2.7/dist-packages/pybombs/packagers/apt.py and removed the reference to "-Y" and that seems to have allowed me to install gnuradio without having all of the package failures and such.

    def install(self, pkgname):
        """
        apt(-get) **-y** install pkgname
        """
        try:
            subproc.monitor_process([self.getcmd, **"-y",** "install", pkgname], elevate=True, throw=True)

I removed the **-y** from the first line referencing it and the **"-y",**  from the second line referencing it.

Don't laugh too hard.   I have no clue what I'm doing.   But it seems to work.

mbr0wn, 2016-10-16:
I backed out apt for now. It even warns not be used in scripts -- I should have heeded the warning.

closed 2016-10-16 by mbr0wn