gnuradio/pybombs #402: Error regarding package installation check when viewing recipes list after adding standard recipes

courtarro, 2016-10-26:
Here's what I'm seeing when I run `pybombs -vv recipes list`. I've tried with `python-apt` installed, and without. I'm running on Ubuntu 14.04.

```
PyBOMBS.Recipe[gr-benchmark] - OBNOXIOUS - Loading recipe file: /home/vagrant/.pybombs/recipes/gr-etcetera/gr-benchmark.lwr
PyBOMBS.Recipe[gr-benchmark] - OBNOXIOUS - Loading template file: /home/vagrant/.local/lib/python2.7/site-packages/pybombs/templates/empty.lwt
PyBOMBS.Recipe[gr-benchmark] - OBNOXIOUS - Inheriting from file /home/vagrant/.local/lib/python2.7/site-packages/pybombs/templates/empty.lwt
PyBOMBS.Recipe[gr-benchmark] - OBNOXIOUS - Loaded recipe - gr-benchmark
.PyBOMBS.PackageManager - DEBUG - Checking if package gr-benchmark is installed...
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (gr-benchmark)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (gr-benchmark)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (gr-benchmark)
PyBOMBS.PackageManager - DEBUG - Package gr-benchmark is requesting a source build.
Traceback (most recent call last):
  File "/home/vagrant/.local/bin/pybombs", line 11, in <module>
    load_entry_point('PyBOMBS==2.2.0', 'console_scripts', 'pybombs')()
  File "/home/vagrant/.local/lib/python2.7/site-packages/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/home/vagrant/.local/lib/python2.7/site-packages/pybombs/commands/base.py", line 188, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/home/vagrant/.local/lib/python2.7/site-packages/pybombs/commands/base.py", line 114, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/home/vagrant/.local/lib/python2.7/site-packages/pybombs/commands/recipes.py", line 218, in run_list
    'installed_by': format_pkg_list(pkgmgr.installed(pkg, return_pkgr_name=True)),
  File "/home/vagrant/.local/lib/python2.7/site-packages/pybombs/package_manager.py", line 163, in installed
    pkg_version = pkgr.installed(r)
AttributeError: 'NoneType' object has no attribute 'installed'

```

courtarro, 2016-10-26:
`git bisect` informs me that this commit introduced the bug, or at least exposed it: 73cfb183f0a144795f7a53a70ffad54ff17478cf

mbr0wn, 2016-10-26:
Thanks for bisecting! This is something we should fix before a 2.3 release.

mbr0wn, 2016-11-17:
Hm, can't repro anymore. Don't really want to go back... does this still not work for you?

closed 2016-11-18 by mbr0wn