gnuradio/pybombs #404: bladeRF recipe fails if cmake is the old version that GNURadio installs from source

courtarro, 2016-10-26:
The `cmake.lwr` states:

    source: wget+http://www.cmake.org/files/v2.8/cmake-2.8.6.tar.gz

Unfortunately, the bladeRF build process includes the `target_compile_definitions` command, which requires cmake 3.0 or newer. Could we bump the version of cmake installed via source? What would be the side effects?

mbr0wn, 2016-10-26:
Yeah, we can do that np. However, what would be tough to do is to bump the min version for the CMake that is *not* installed from source. So if you're on Ubuntu 14.04, cmake will, by default, not install from source and you're stuck with the old version. Don't have a good workaround for that right now. We could make a separate cmake-3 recipe...?

courtarro, 2016-10-26:
Another data point: I set up a Vagrant box running `bento/ubuntu-14.04`. With a fresh install of PyBOMBS, I told it to install bladeRF and I got no errors at all. The installed version of cmake, from apt, is `2.8.12.2`. Doing a bit more testing now...

courtarro, 2016-10-26:
Confirmed that the `target_compile_definitions` feature was added in cmake `2.8.11`. This version is included in 14.04, but the latest available package in 12.04 is cmake `2.8.7`. So basically the bladeRF source installation doesn't support `Precise` out of the box.

mbr0wn, 2016-10-26:
We could make 2.8.11 the required minimum. I'd be fine with that. It would mean updating the cmake.lwr to check for that version; on Precise, it would then not find a new-enough CMake and build from source instead.

mbr0wn, 2016-10-26:
As a side note, we can have pretty much any version of CMake be the source version, as long as it's not too new and then breaks other things. If it's backward compatible, we should probably just fast-forward it to whatever the newest version is.

Do you know if CMake requires any bootstrapping software? Or can it build itself? (e.g., autotools does not work if there is no older version of autotools installed)

courtarro, 2016-10-26:
Currently working on that. There are issues with `Precise`: we can't download from `cmake.org` because their HTTPS certificate is too fancy. Switching to Git. Using Git to clone `v2.8.12.2` (latest from v2) or `v3.6.2` (the latest mature release), but the included `configure` script doesn't like the `CC=` option we're passing it.

mbr0wn, 2016-10-26:
Those `CC=` options are breaking my cross compiles too and I'm fed up with them. @osh, sorry, they need to go -- I'll promise I won't take away the ability to choose your compiler.

mbr0wn, 2016-11-18:
Updated to 3.7.

mbr0wn, 2016-12-01:
@courtarro What do you think about the current state? We have a couple of options:
- Close this and move on
- Mandate CMake 3.1 for everyone. I think that's unreasonable though.
- Make a separate recipe for cmake 3.1 (and call it cmake3 or something like that). Add a dependency onto that for bladeRF and whoever else needs it. If cmake3 is installed, checks for cmake will automatically pass.

mbr0wn, 2016-12-01:
I'm leaning towards the last one btw.

mbr0wn, 2016-12-01:
I mixed up things along the way. If we need 2.8.11, and that's in 14.04, I think we can mandate that as a new minimum. Older distros will still work, they just need to install CMake from source. So let's bump that, and close this issue.

closed 2016-12-01 by mbr0wn