gnuradio/pybombs #406: error

sincos364, 2016-10-27:
sudo pip install --upgrade git+https://github.com/gnuradio/pybombs.git
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 564, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2608, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2268, in load
    return self.resolve()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2274, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead

mbr0wn, 2016-10-31:
Which system is this on?

sincos364, 2016-10-31:
Debian for raspberry pi 3

mbr0wn, 2016-10-31:
So this is happening in `mercurial.py` from pip, which we don't use. It's possible that some setting we're setting is triggering this, but most of what we do is pretty standard. This might be a bug in pip, actually -- wouldn't be the first one.

sincos364, 2016-10-31:
Del pip

mbr0wn, 2016-10-31:
You can install PyBOMBS without pip. git clone and then use setuptools.

sincos364, 2016-11-01:
pi@raspberrypi:~ $ sudo pip install PyBOMBS
Downloading/unpacking PyBOMBS
  Downloading PyBOMBS-2.2.0.tar.gz (65kB): 65kB downloaded
  Running setup.py (path:/tmp/pip-build-2LVjDE/PyBOMBS/setup.py) egg_info for package PyBOMBS
    
Requirement already satisfied (use --upgrade to upgrade): setuptools in /usr/lib/python2.7/dist-packages (from PyBOMBS)
Downloading/unpacking PyYAML (from PyBOMBS)
  Downloading PyYAML-3.12.tar.gz (253kB): 253kB downloaded
  Running setup.py (path:/tmp/pip-build-2LVjDE/PyYAML/setup.py) egg_info for package PyYAML
    
Requirement already satisfied (use --upgrade to upgrade): requests in /usr/lib/python2.7/dist-packages (from PyBOMBS)
Requirement already satisfied (use --upgrade to upgrade): six in /usr/lib/python2.7/dist-packages (from PyBOMBS)
Downloading/unpacking future (from PyBOMBS)
  Downloading future-0.16.0.tar.gz (824kB): 824kB downloaded
  Running setup.py (path:/tmp/pip-build-2LVjDE/future/setup.py) egg_info for package future
    
    warning: no files found matching '*.au' under directory 'tests'
    warning: no files found matching '*.gif' under directory 'tests'
    warning: no files found matching '*.txt' under directory 'tests'
Installing collected packages: PyBOMBS, PyYAML, future
  Running setup.py install for PyBOMBS
    
    Installing pybombs script to /usr/local/bin
  Running setup.py install for PyYAML
    checking if libyaml is compilable
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c build/temp.linux-armv7l-2.7/check_libyaml.c -o build/temp.linux-armv7l-2.7/check_libyaml.o
    build/temp.linux-armv7l-2.7/check_libyaml.c:2:18: fatal error: yaml.h: Нет такого файла или каталога
     #include <yaml.h>
                      ^
    compilation terminated.
    
    libyaml is not found or a compiler error: forcing --without-libyaml
    (if libyaml is installed correctly, you may need to
     specify the option --include-dirs or uncomment and
     modify the parameter include_dirs in setup.cfg)
    
  Running setup.py install for future
    
    warning: no files found matching '*.au' under directory 'tests'
    warning: no files found matching '*.gif' under directory 'tests'
    warning: no files found matching '*.txt' under directory 'tests'
    Installing pasteurize script to /usr/local/bin
    Installing futurize script to /usr/local/bin
Successfully installed PyBOMBS PyYAML future
Cleaning up...

mbr0wn, 2016-11-01:
OK looks like that's working.

closed 2016-11-01 by mbr0wn