gnuradio/pybombs #407: After git pull, pybombs update sais "Error updating package autotools-bootstrap. Aborting."

u0m3, 2016-11-03:
Hello.

I just updated to the latest version of PyBOMBS and I keep getting an error about not being able to update autotools-bootstrap
```PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Updating package: autotools-bootstrap
PyBOMBS.Packager.source - WARNING - Cannot find a source URI for package autotools-bootstrap
PyBOMBS.install_manager - ERROR - Error updating package autotools-bootstrap. Aborting.
```
Now as far as I can deduct from the recipe it should not try to update it, it should just check to see if it exists.

mbr0wn, 2016-11-04:
You'll need to provide some more detail (what are you doing, what's your system...).

u0m3, 2016-11-06:
I apologize, it slipped my mind.
The command I ran was `pybombs update` (after sourcing my _setup_env.sh_ for my env) and the output was what I showed above (minus 42k of lines tree)
```
rpopescu@thor:~$ pybombs update
PyBOMBS - INFO - PyBOMBS Version 2.2.0
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
You are using pip version 8.1.1, however version 8.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Install tree:
[... 42k lines of tree ...]
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Updating package: autotools-bootstrap
PyBOMBS.Packager.source - WARNING - Cannot find a source URI for package autotools-bootstrap
PyBOMBS.install_manager - ERROR - Error updating package autotools-bootstrap. Aborting.
```
My operating system is **Ubuntu 16.04.1 LTS**

mbr0wn, 2016-11-17:
Confirmed.

mbr0wn, 2016-11-17:
0b780a384cfa495d4f00f9f55cd0925f68be9d49

closed 2016-11-17 by mbr0wn