gnuradio/pybombs #414: ERROR - Package has no install method: ssl

cjw0dakuan, 2016-12-15:
I installed pybombs successfully, but when I used the command "pybombs prefix init ~/prefix -a myprefix -R gnuradio-default" to install gnuradio, I met the error "PyBOMBS.DepManager - ERROR - Package has no install method: ssl (declared as dependency for package libevent)" in the Phase 1 "Creating install tree and installing binary packages". How can I do?

mbr0wn, 2016-12-15:
Which distro are you on?

cjw0dakuan, 2016-12-15:
My linux is Centos 7 x86_64-bit

mbr0wn, 2016-12-15:
You're not the first to report this, but I can't reproduce it (I use a vanilla Centos7 Docker image to test that distro). What happens if you run

    $ pybombs -vv install ssl

(can you post the log plz)?

mbr0wn, 2016-12-15:
Just re-confirmed that a vanilla docker image will properly install this.

mbr0wn, 2016-12-15:
Also, does your system actually have openssl-devel:
```
[root@9177e3ddb108 /]# yum search openssl-dev
Loaded plugins: fastestmirror, ovl
base                                                                                                                                                   | 3.6 kB  00:00:00     
extras                                                                                                                                                 | 3.4 kB  00:00:00     
updates                                                                                                                                                | 3.4 kB  00:00:00     
(1/4): base/7/x86_64/group_gz                                                                                                                          | 155 kB  00:00:00     
(2/4): extras/7/x86_64/primary_db                                                                                                                      | 166 kB  00:00:00     
(3/4): updates/7/x86_64/primary_db                                                                                                                     | 1.2 MB  00:00:01     
(4/4): base/7/x86_64/primary_db                                                                                                                        | 5.6 MB  00:00:04     
Determining fastest mirrors
 * base: centos.host-engine.com
 * extras: centos.sonn.com
 * updates: mirror.raystedman.net
========================================================================== N/S matched: openssl-dev ==========================================================================
openssl-devel.i686 : Files for development of applications which will use OpenSSL
openssl-devel.x86_64 : Files for development of applications which will use OpenSSL
xmlsec1-openssl-devel.i686 : OpenSSL crypto plugin for XML Security Library
xmlsec1-openssl-devel.x86_64 : OpenSSL crypto plugin for XML Security Library
```

...do you get the same output?

cjw0dakuan, 2016-12-15:
please wait a moment, i am sorry but i am busy for something...,

cjw0dakuan, 2016-12-15:
i run pybombs -vv install ssl, the result is:

#  pybombs -vv install ssl
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /root/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {'default_prefix': 'myprefix'}
PyBOMBS.ConfigManager - DEBUG - Template directory: /usr/lib/python2.7/site-packages/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /root/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias `myprefix'.
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/root/prefix)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /root/prefix/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /root/prefix/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /root/prefix/src
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix inventory file not found: /root/prefix/.pybombs/inventory.yml
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /root/prefix/.pybombs/inventory.yml...
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /root/prefix/.pybombs/recipes
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /root/prefix/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/root/prefix/.pybombs/recipes', '/root/.pybombs/recipes/gr-etcetera', '/root/.pybombs/recipes/gr-recipes', '/usr/lib/python2.7/site-packages/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBOMBS - INFO - PyBOMBS Version 2.3.0
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /root/prefix/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/root/prefix/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/bjam.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/setuptools.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/bootstrapautoconf.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/empty.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/prefix.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/distutils.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/boostflags.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/sdk.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/cmake.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/lib/python2.7/site-packages/pybombs/templates/autoconf.lwt
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /root/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/root/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 71 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /root/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/root/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 166 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/lib/python2.7/site-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/lib/python2.7/site-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.install - DEBUG - Initializing command class for command install
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - yumdnf is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
[... cut ...]
PyBOMBS.PackageManager - DEBUG - install(ssl, install_type=source, static=False)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (ssl)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (ssl)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (ssl)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (ssl)
PyBOMBS.PackageManager - DEBUG - Using packager source
PyBOMBS.Packager.source - WARNING - Cannot find a source URI for package ssl
PyBOMBS.install_manager - ERROR - Error installing package ssl. Aborting.

cjw0dakuan, 2016-12-15:
i run yum search openssl-dev, the result is:

yum search openssl-dev
Loaded plugins:：fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: centos.ustc.edu.cn
 * epel: mirror01.idc.hinet.net
 * extras: centos.ustc.edu.cn
 * updates: centos.ustc.edu.cn
=========================== N/S matched: openssl-dev ===========================
openssl-devel.i686 : Files for development of applications which will use
                   : OpenSSL
openssl-devel.x86_64 : Files for development of applications which will use
                     : OpenSSL
xmlsec1-openssl-devel.x86_64 : OpenSSL crypto plugin for XML Security Library
xmlsec1-openssl-devel.i686 : OpenSSL crypto plugin for XML Security Library

mbr0wn, 2016-12-15:
Can you please also post the output of

    $ yum info openssl-devel

Thanks!

The issue is here:
```
PyBOMBS.PackageManager - DEBUG - Using packager yumdnf
PyBOMBS.Packager.yumdnf - OBNOXIOUS - install(ssl, static=False)
PyBOMBS.ReqScanner - OBNOXIOUS - Adding package with name openssl-devel
PyBOMBS.ReqScanner - OBNOXIOUS - End of requirements list
PyBOMBS.ReqScanner - OBNOXIOUS - Done parsing requirements string openssl-devel
PyBOMBS.Packager.yumdnf - OBNOXIOUS - Calling ev for recursive satisfier rule evaluation
PyBOMBS.Packager.yumdnf - DEBUG - Package openssl-devel has version False in yum
```
The last line tells me it can't even determine a version, but yum clearly has the packet (see your previous post).

cjw0dakuan, 2016-12-15:
The output is :
# yum info openssl-devel
已加载插件：fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: centos.ustc.edu.cn
 * epel: mirror01.idc.hinet.net
 * extras: centos.ustc.edu.cn
 * updates: centos.ustc.edu.cn
可安装的软件包
名称    ：openssl-devel
架构    ：i686
时期       ：1
版本    ：1.0.1e
发布    ：60.el7
大小    ：1.2 M
源    ：base/7/x86_64
简介    ： Files for development of applications which will use OpenSSL
网址    ：http://www.openssl.org/
协议    ： OpenSSL
描述    ： OpenSSL is a toolkit for supporting cryptography. The openssl-devel
         : package contains include files needed to develop applications which
         : support various cryptographic algorithms and protocols.

名称    ：openssl-devel
架构    ：x86_64
时期       ：1
版本    ：1.0.1e
发布    ：60.el7
大小    ：1.2 M
源    ：base/7/x86_64
简介    ： Files for development of applications which will use OpenSSL
网址    ：http://www.openssl.org/
协议    ： OpenSSL
描述    ： OpenSSL is a toolkit for supporting cryptography. The openssl-devel
         : package contains include files needed to develop applications which
         : support various cryptographic algorithms and protocols.

mbr0wn, 2016-12-15:
Ah, I see. The problem is that PyBOMBS doesn't handle internationalization. It's expecting English output.

mbr0wn, 2016-12-15:
Can you try

    $ LC_ALL=C yum info openssl-devel

?

cjw0dakuan, 2016-12-15:
The output is:
LC_ALL=C yum info openssl-devel
Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: centos.ustc.edu.cn
 * epel: mirror01.idc.hinet.net
 * extras: centos.ustc.edu.cn
 * updates: centos.ustc.edu.cn
Available Packages
Name        : openssl-devel
Arch        : i686
Epoch       : 1
Version     : 1.0.1e
Release     : 60.el7
Size        : 1.2 M
Repo        : base/7/x86_64
Summary     : Files for development of applications which will use OpenSSL
URL         : http://www.openssl.org/
License     : OpenSSL
Description : OpenSSL is a toolkit for supporting cryptography. The
            : openssl-devel package contains include files needed to develop
            : applications which support various cryptographic algorithms and
            : protocols.

Name        : openssl-devel
Arch        : x86_64
Epoch       : 1
Version     : 1.0.1e
Release     : 60.el7
Size        : 1.2 M
Repo        : base/7/x86_64
Summary     : Files for development of applications which will use OpenSSL
URL         : http://www.openssl.org/
License     : OpenSSL
Description : OpenSSL is a toolkit for supporting cryptography. The
            : openssl-devel package contains include files needed to develop
            : applications which support various cryptographic algorithms and
            : protocols.

mbr0wn, 2016-12-15:
OK, try updating PyBOMBS to the latest git version

    $     $ pip install --upgrade git+https://github.com/gnuradio/pybombs.git

mbr0wn, 2016-12-15:
might work now.

cjw0dakuan, 2016-12-15:
it is very slow, when i execute the command...
pip install --upgrade git+https://github.com/gnuradio/pybombs.git
Collecting git+https://github.com/gnuradio/pybombs.git
  Cloning https://github.com/gnuradio/pybombs.git to /tmp/pip-9riYUX-build

-----------------------------------------
but i download the pybombs-master.zip very fast. Can I use the zip to install pybombs?How can I do?

cjw0dakuan, 2016-12-15:
Thank you for your help, I solved the error, but I met another error in the Phase 2, the error is 
"Install tree:
|
\- gnuradio
   |
   +- uhd
   |
   +- apache-thrift
   |  |
   |  \- automake
   |
   \- pyqwt5
      |
      \- automake
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: automake
02245 kB / 02245 kB (100%)
Configuring: (100%) [=========================================================]
Building:    (100%) [=========================================================]
Installing:  (100%) [=========================================================]
PyBOMBS.install_manager - INFO - Installation successful.
PyBOMBS.install_manager - INFO - Installing package: pyqwt5
PyBOMBS.Fetcher - ERROR - Unexpected error while fetching wget+http://sourceforge.net/projects/pyqwt/files/pyqwt5/PyQwt-5.2.0/PyQwt-5.2.0.tar.gz.
PyBOMBS.Fetcher - ERROR - 'X509' object has no attribute '_x509'
PyBOMBS.Packager.source - ERROR - Problem occurred while building package pyqwt5:
Unable to fetch recipe pyqwt5
PyBOMBS.install_manager - ERROR - Error installing package pyqwt5. Aborting.
"

mbr0wn, 2016-12-15:
This is something else. Closing. Please bring this up on the mailing list. Maybe you can propose a fix given my previous changes.

closed 2016-12-15 by mbr0wn