gnuradio/pybombs #416: gr-osmosdr builds a lib that depends on libgnuradio-runtime.so.3.7.9 of the host system

hackwerk, 2016-12-20:
When I build gr-osmosdr (pybombs install gr-osmosdr) on top of a prefix with gnuradio-default installed, I get a libgnuradio-osmosdr that depends on the gnuradio libs in the host system instead of the pybombs prefix.

Note that for instance both libgnuradio-runtime.so.3.7.9 and libgnuradio-runtime-3.7.11git.so.0.0.0 are referenced, where 3.7.9 is the host system and 3.7.11git is in the pybombs prefix folder.

Is this expected?
Am I making a mistake somewhere?

It causes segfaults in gqrx which depends on gr-osmosdr.

```
~> ldd lib/libgnuradio-osmosdr.so
        linux-vdso.so.1 =>  (0x00007f3cb3a60000)
        libboost_thread.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0 (0x00007f3cb3514000)
        libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007f3cb330f000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f3cb30f2000)
        libgnuradio-runtime-3.7.11git.so.0.0.0 => /home/pybombs/gnuradio-pybombs/base/lib/libgnuradio-runtime-3.7.11git.so.0.0.0 (0x00007f3cb2d78000)
        libgnuradio-pmt-3.7.11git.so.0.0.0 => /home/pybombs/gnuradio-pybombs/base/lib/libgnuradio-pmt-3.7.11git.so.0.0.0 (0x00007f3cb2b28000)
        libgnuradio-blocks-3.7.11git.so.0.0.0 => /home/pybombs/gnuradio-pybombs/base/lib/libgnuradio-blocks-3.7.11git.so.0.0.0 (0x00007f3cb2633000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f3cb242b000)
        libgnuradio-iqbalance.so => /home/pybombs/gnuradio-pybombs/base/lib/libgnuradio-iqbalance.so (0x00007f3cb2219000)
        libosmosdr.so.0 => /home/pybombs/gnuradio-pybombs/base/lib/libosmosdr.so.0 (0x00007f3cb2014000)
        libgnuradio-fcd-3.7.11git.so.0.0.0 => /home/pybombs/gnuradio-pybombs/base/lib/libgnuradio-fcd-3.7.11git.so.0.0.0 (0x00007f3cb1dfc000)
        libgnuradio-fcdproplus.so.0 => /usr/lib/x86_64-linux-gnu/libgnuradio-fcdproplus.so.0 (0x00007f3cb1be6000)
        librtlsdr.so.0 => /home/pybombs/gnuradio-pybombs/base/lib/librtlsdr.so.0 (0x00007f3cb19d7000)
        libgnuradio-uhd-3.7.11git.so.0.0.0 => /home/pybombs/gnuradio-pybombs/base/lib/libgnuradio-uhd-3.7.11git.so.0.0.0 (0x00007f3cb1786000)
        libuhd.so.3 => /home/pybombs/gnuradio-pybombs/base/lib/libuhd.so.3 (0x00007f3cb0c3c000)
        libhackrf.so.0 => /home/pybombs/gnuradio-pybombs/base/lib/libhackrf.so.0 (0x00007f3cb0a36000)
        libbladeRF.so.1 => /home/pybombs/gnuradio-pybombs/base/lib/libbladeRF.so.1 (0x00007f3cb0804000)
        libairspy.so.0 => /home/pybombs/gnuradio-pybombs/base/lib/libairspy.so.0 (0x00007f3cb05fb000)
        libSoapySDR.so.0.6-dev => /usr/local/lib/libSoapySDR.so.0.6-dev (0x00007f3cb03bc000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f3cb003a000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f3cafd30000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f3cafb1a000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f3caf751000)
        /lib64/ld-linux-x86-64.so.2 (0x000055dce0375000)
        libthrift-0.9.3.so => /home/pybombs/gnuradio-pybombs/base/lib/libthrift-0.9.3.so (0x00007f3caf41b000)
        libvolk.so.1.3 => /home/pybombs/gnuradio-pybombs/base/lib/libvolk.so.1.3 (0x00007f3caf050000)
        libboost_program_options.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.58.0 (0x00007f3caedd2000)
        libboost_filesystem.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0 (0x00007f3caebb9000)
        liblog4cpp.so.5 => /usr/lib/liblog4cpp.so.5 (0x00007f3cae97c000)
        libosmodsp.so.0 => /home/pybombs/gnuradio-pybombs/base/lib/libosmodsp.so.0 (0x00007f3cae776000)
        libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f3cae55e000)
        libgnuradio-audio-3.7.11git.so.0.0.0 => /home/pybombs/gnuradio-pybombs/base/lib/libgnuradio-audio-3.7.11git.so.0.0.0 (0x00007f3cae324000)
        libhidapi-libusb.so.0 => /usr/lib/x86_64-linux-gnu/libhidapi-libusb.so.0 (0x00007f3cae11c000)
        libgnuradio-runtime.so.3.7.9 => /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.9 (0x00007f3cade22000)
        libgnuradio-blocks.so.3.7.9 => /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so.3.7.9 (0x00007f3cad93b000)
        libgnuradio-audio.so.3.7.9 => /usr/lib/x86_64-linux-gnu/libgnuradio-audio.so.3.7.9 (0x00007f3cad6eb000)
        libgnuradio-pmt.so.3.7.9 => /usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so.3.7.9 (0x00007f3cad499000)  
        libboost_date_time.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.58.0 (0x00007f3cad287000)
        libboost_regex.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0 (0x00007f3cacf7f000)
        libboost_serialization.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.58.0 (0x00007f3cacd1e000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f3cacb19000)
        libssl.so.1.0.0 => /lib/x86_64-linux-gnu/libssl.so.1.0.0 (0x00007f3cac8b0000)
        libcrypto.so.1.0.0 => /lib/x86_64-linux-gnu/libcrypto.so.1.0.0 (0x00007f3cac46b000)
        liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007f3cac1eb000)
        libfftw3f.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f.so.3 (0x00007f3cabddf000)
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f3cabdbf000)
        libasound.so.2 => /usr/lib/x86_64-linux-gnu/libasound.so.2 (0x00007f3cababf000)
        libjack.so.0 => /usr/lib/x86_64-linux-gnu/libjack.so.0 (0x00007f3cab89e000)
        libvolk.so.1.1 => /usr/lib/x86_64-linux-gnu/libvolk.so.1.1 (0x00007f3cab4e7000)
        libportaudio.so.2 => /usr/lib/x86_64-linux-gnu/libportaudio.so.2 (0x00007f3cab2b6000)
        libicui18n.so.55 => /usr/lib/x86_64-linux-gnu/libicui18n.so.55 (0x00007f3caae54000)
        libicuuc.so.55 => /usr/lib/x86_64-linux-gnu/libicuuc.so.55 (0x00007f3caaabf000)
        libdb-5.3.so => /usr/lib/x86_64-linux-gnu/libdb-5.3.so (0x00007f3caa712000)
        libicudata.so.55 => /usr/lib/x86_64-linux-gnu/libicudata.so.55 (0x00007f3ca8c5a000)

```

hackwerk, 2016-12-20:
I just realized I wanted to put this in gr-recipes. Please close if misplaced.

mbr0wn, 2017-01-04:
Yeah this was discussed here: https://github.com/gnuradio/gr-recipes/issues/69

closed 2017-01-04 by mbr0wn