gnuradio/pybombs #417: Import Error: cannot import name qtgui

james-tate, 2016-12-22:
Running Pybombs on fresh OS install. When starting gnuradio-companion the qt gui widgets do not import. When trying to run the default .grc which gets generated when starting gnuradio-companion I get an import error. 

top_block.py", line 26, in <module>
    from gnuradio import qtgui
ImportError: cannot import name qtgui

james-tate, 2016-12-23:
It appears Pybombs does not handle the installation of QT4 and QWT. I ran Synaptic and searched for QWT. This installed QT4 as a prereq. I then ran "pybombs update gnuradio" and it enabled gr-qtgui support. In retrospect I guess pybombs rebuild would have worked as well.

closed 2016-12-28 by james-tate

mbr0wn, 2017-01-03:
PyBOMBS *should* handle installing those though.

james-tate, 2017-01-04:
Can confirm it does not seem to install gr-qtgui. I have tried on 2 separate machines using fresh installations of Mint 18.

jahagirdar, 2019-02-07:
Seeing the same issue on centos 7

tuf22191, 2019-11-22:
Hello resolution?
I am seeing this on Centos 7 too