gnuradio/pybombs #418: PyBOMBS Package has no install method: wget

CaptainMcCrank, 2016-12-28:
I'm trying to install gr-gsm, and I've run into a roadblock involving pybombs:

sudo pybombs install gr-gsm
PyBOMBS - INFO - PyBOMBS Version 2.3.1a
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.install_manager - ERROR - Package has no install method: wget

Any advice on how to proceed?

james-tate, 2016-12-28:
Hi CaptainMcCrank,

I had some issues with PyBOMBS installing prerequisites earlier this week. I used synaptic to pull down whatever PyBOMBS was leaving out.  You can use "sudo apt-get install synaptic" to install, then run using "sudo synaptic". Finally, use the search to find wget, mark it for installation, then click apply. After that you should be able to run "pybombs install gr-gsm" and see what happens.

CaptainMcCrank, 2016-12-28:
Thanks James.  I meant to mention that I'm on an OSX machine- so I'm not sure that updating synaptic is going to work for me. 

Wget is installed on the machine.  In looking at the error- I'm guessing that pybombs isn't seeing a configuration for a "make install" type of event in gr-recipes or gr-etcetera... but since I don't really understand how pybombs works, maybe I'm misunderstanding things.

mbr0wn, 2017-01-03:
Are your recipes up to date?
```
$ pybombs recipes update
```

What's odd here is that while there is a package called 'wget', no one actually requires it. Weird. Posting the output of

```
$ pybombs -vv install gr-gsm
```

could help us, too. Thanks!

CaptainMcCrank, 2017-01-04:
Recipes were up to date.  Here's the output of pybombs -vv install gr-gsm: 

PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /Users/captainmccrank/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {'default_prefix': 'default_prx'}
PyBOMBS.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/site-packages/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /Users/captainmccrank/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias `default_prx'.
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/usr/local)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /usr/local/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /usr/local/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /usr/local/src
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix inventory file not found: /usr/local/.pybombs/inventory.yml
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /usr/local/.pybombs/inventory.yml...
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /usr/local/.pybombs/recipes
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /usr/local/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/usr/local/.pybombs/recipes', '/Users/captainmccrank/.pybombs/recipes/gr-etcetera', '/Users/captainmccrank/.pybombs/recipes/gr-recipes', '/usr/local/lib/python2.7/site-packages/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBOMBS - INFO - PyBOMBS Version 2.3.1a
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/autoconf.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/bjam.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/boostflags.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/bootstrapautoconf.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/cmake.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/distutils.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/empty.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/prefix.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/sdk.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /usr/local/lib/python2.7/site-packages/pybombs/templates/setuptools.lwt
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/captainmccrank/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/captainmccrank/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 71 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/captainmccrank/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/captainmccrank/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 167 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/site-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/site-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.install - DEBUG - Initializing command class for command install
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - brew is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
[... cut ...]
PyBOMBS.Packager.brew - OBNOXIOUS - exists(python-zmq)
PyBOMBS.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBOMBS.Packager.pkgconfig - OBNOXIOUS - exists(python-zmq)
PyBOMBS.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBOMBS.Packager.cmd - OBNOXIOUS - exists(python-zmq)
PyBOMBS.ReqScanner - OBNOXIOUS - Empty requirements string.
PyBOMBS.PackageManager - DEBUG - Package python-zmq is not installable.
PyBOMBS.install_manager - ERROR - Package has no install method: python-zmq
PyBOMBS - DEBUG - Unresolved install path.

mbr0wn, 2017-01-04:
This log is an entirely different issue from what you originally posted. What system are you on? Maybe Mac? Does this look familiar: #419 (see towards the end).

mbr0wn, 2017-07-09:
I'm not clear if this is still an issue. Closing now. Feel free to reopen if there's more input.

closed 2017-07-09 by mbr0wn

mluis, 2017-10-05:
I'm having the same issue

mluis, 2017-10-05:
Solved it. I was missing installing automake 👍

mluis, 2017-10-05:
now I have
`ERROR - Package has no install method: python`

ramgarden, 2018-07-12:
I'm also getting Package has no install method: python.
I'm using Chocolatey on Windows 10.
I was able to get python and python3 installed, also got pybombs installed OK.

saintofinternet, 2020-01-03:
``
The following NEW packages will be installed:
  python-zmq
0 upgraded, 1 newly installed, 0 to remove and 35 not upgraded.
Need to get 267 kB of archives.
After this operation, 1,350 kB of additional disk space will be used.
Get:1 http://ftp.harukasan.org/kali kali-rolling/main amd64 python-zmq amd64 17.1.2-4 [267 kB]
Fetched 267 kB in 15s (17.2 kB/s)                                                                               
Selecting previously unselected package python-zmq.
(Reading database ... 378165 files and directories currently installed.)
Preparing to unpack .../python-zmq_17.1.2-4_amd64.deb ...
Unpacking python-zmq (17.1.2-4) ...
Setting up python-zmq (17.1.2-4) ...
**PyBOMBS.install_manager - ERROR - Package has no install method: python-click-plugins**
``
Same issue with my....  using Kali Virtual Box Image running on Oracle VB 6.1 with Windows 7 Host.

velichkov, 2020-01-03:
Hi @saintofinternet,

> **PyBOMBS.install_manager - ERROR - Package has no install method: python-click-plugins**`
> 
> Same issue with me.... using Kali Virtual Box Image running on Oracle VB 6.1 with Windows 7 Host.

Your issue is not exactly the same, it's about `python-click-plugins` and not `wget`

It seems in Debian Testing and Kali they have started removing some python2 packages as python2 is no longer maintained, see https://pythonclock.org/

So my advice is reinstall pybombs with pip3 in order to use python3, see https://github.com/gnuradio/pybombs/#using-python3
```
sudo pip uninstall pybombs
sudo apt install python3-pip
sudo pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git
```