gnuradio/pybombs #419: PyBOMBS.DepManager - ERROR - Package has no install method: ssl

stuzog, 2017-01-01:
I'm having the error come up with trying to install gnuradio on OS X 10.11 after a basic install of pyBOMBS using `sudo pip install [--upgrade] git+https://github.com/gnuradio/pybombs.git`:

```
PyBOMBS.DepManager - ERROR - Package has no install method: ssl (declared as dependency for package lib event)
```

My ~/.pybombs/config.yaml:

config:
  default_prefix: gnuradio
prefix_aliases:
  gnuradio: /Users/Stuart/gnuradio
  myprefix: /Users/Stuart/gnuradio
recipes:
  gr-etcetera: git+https://github.com/gnuradio/gr-etcetera.git
  gr-recipes: git+https://github.com/gnuradio/gr-recipes.git

I ran the recipe list as above:

Last login: Sat Dec 31 16:15:05 on ttys001
You have mail.
-bash: uname: command not found
-bash: ps: command not found
bash$ pybombs -v recipes list -l "^python$"
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /Users/Stuart/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {'default_prefix': 'gnuradio'}
PyBOMBS.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/site-packages/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /Users/Stuart/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias `gnuradio'.
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/Users/Stuart/gnuradio)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /Users/Stuart/gnuradio/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /Users/Stuart/gnuradio/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /Users/Stuart/gnuradio/src
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix inventory file not found: /Users/Stuart/gnuradio/.pybombs/inventory.yml
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /Users/Stuart/gnuradio/.pybombs/inventory.yml...
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /Users/Stuart/gnuradio/.pybombs/recipes
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /Users/Stuart/gnuradio/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/Users/Stuart/gnuradio/.pybombs/recipes', '/Users/Stuart/.pybombs/recipes/gr-etcetera', '/Users/Stuart/.pybombs/recipes/gr-recipes', '/usr/local/lib/python2.7/site-packages/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBOMBS - INFO - PyBOMBS Version 2.3.0
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/Stuart/gnuradio/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/Stuart/gnuradio/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/Stuart/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/Stuart/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 71 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/Stuart/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/Stuart/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 166 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/site-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/site-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.recipes - DEBUG - Initializing command class for command recipes
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - brew is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using binary packagers: ['pip', 'brew', 'pkgconfig', 'cmd']
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/Stuart/gnuradio/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/Stuart/gnuradio/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/Stuart/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/Stuart/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 71 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /Users/Stuart/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/Users/Stuart/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 166 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/site-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/site-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.recipes - DEBUG - Loading all package names
Loading package information....PyBOMBS.PackageManager - DEBUG - Checking if package python is installed...
PyBOMBS.Packager.cmd - DEBUG - Found version number: 2.7.12
PyBOMBS.PackageManager - DEBUG - Checking if package python is installable...
PyBOMBS.Packager.cmd - DEBUG - Found version number: 2.7.12

Package Name  Recipe Filename                           Installed By  
----------------------------------------------------------------------
python        ~/.pybombs/recipes/gr-recipes/python.lwr  cmd     
Any suggestions?

mbr0wn, 2017-01-03:
This looks a lot like #414. What about

```
$ pybombs recipes list -l ssl
```

?

stuzog, 2017-01-03:
OK, looks like there was a problem as to which python was being used per .bash_profile paths. After re-arranging this and installing some required packages, everything works:

```
bash$ pybombs -v recipes list -l "^python$"
(....mucho pretty coloured output...)
Package Name  Recipe Filename                           Installed By  
----------------------------------------------------------------------
python        ~/.pybombs/recipes/gr-recipes/python.lwr  cmd           

bash$ pybombs recipes list -l ssl
PyBOMBS - INFO - PyBOMBS Version 2.3.1a
Loading package information.....
Package Name  Recipe Filename                                Installed By  
---------------------------------------------------------------------------
libpolarssl   ~/.pybombs/recipes/gr-recipes/libpolarssl.lwr  -             
ssl           ~/.pybombs/recipes/gr-recipes/ssl.lwr          pkgconfig     
```
Looks like it's working -- thanks!

stuzog, 2017-01-03:
However, trying to install gnu radio after removing ~/gnuradio from previous attempts I get the following:

```
bash$ pybombs prefix init ~/gnuradio -a gnuradio -R gnuradio-default
PyBOMBS - INFO - PyBOMBS Version 2.3.1a
PyBOMBS.prefix - INFO - Creating directory `/Users/Stuart/gnuradio'
Alias `gnuradio' already exists, overwrite Y/[N]? Y
PyBOMBS.ConfigManager - INFO - Creating new config file /Users/Stuart/gnuradio/.pybombs/config.yml
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
PyBOMBS.install_manager - ERROR - Package has no install method: python-zmq
bash$ 

```

mbr0wn, 2017-01-03:
What does
```
pybombs -v install python-zmq
```
show? I think the recipe might be outdated. I guess you're using `port` to install stuff, correcto?

stuzog, 2017-01-03:
```
bash$ pybombs -v install python-zmq
(...mucho coloured text...)
PyBOMBS.PackageManager - DEBUG - Using binary packagers: ['pip', 'brew', 'pkgconfig', 'cmd']
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.PackageManager - DEBUG - Checking if package python-zmq is installable...
PyBOMBS.PackageManager - DEBUG - Package python-zmq is not installable.
PyBOMBS.install_manager - ERROR - Package has no install method: python-zmq
PyBOMBS - DEBUG - Unresolved install path.
```

stuzog, 2017-01-03:
Using Homebrew (unfortunately).

I tried installing `pyzmq` as there doesn't seem to be a python-zmq for 
OS X (?), and got this:

```
bash$ sudo pip install pyzmq
Password:
The directory '/Users/Stuart/Library/Caches/pip/http' or its parent 
directory is not owned by the current user and the cache has been 
disabled. Please check the permissions and owner of that directory. If 
executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/Stuart/Library/Caches/pip' or its parent directory 
is not owned by the current user and caching wheels has been disabled. 
check the permissions and owner of that directory. If executing pip with 
sudo, you may want sudo's -H flag.
Collecting pyzmq
Downloading pyzmq-16.0.2-cp27-cp27m-macosx_10_6_intel.whl (1.1MB)
100% |████████████████████████████████| 1.1MB 666kB/s
Installing collected packages: pyzmq
Successfully installed pyzmq-16.0.2
```

mbr0wn, 2017-01-03:
The whole Mac ecosystem is probably the least developed in PyBOMBS, so sorry for any trouble, but it's still pretty new.

The python-zmq recipe doesn't have a line for homebrew. So even if you install it via homebrew, it won't know that you've installed it. This isn't a trivial fix, although I have an idea on how to do that (we could search for the module).

A workaround for you could be to make python-zmq optional:
```
$ pybombs config --package python-zmq optional True
```
It'll try to install, fail, continue, but GNU Radio will probably still find it because you do, in fact, have it installed.

stuzog, 2017-01-03:
OK, but where is `python-zmq` installed?
There's nothing in `~/gnuradio/source/` (`~/gnuradio` is my `~/prefix` 
directory).

mbr0wn, 2017-01-03:
It does not have a source candidate. Your homebrew install will put it wherever it puts Python modules.

mbr0wn, 2017-01-03:
This is turning into a support discussion -- I'll close the issue for now. Please ask on discuss-gnuradio if you want to continue the discussion. Also, thanks for all the reports. I do want to get the Mac stuff more stable, but as I'm not a Mac user myself I'm relying a lot on reports from others.

closed 2017-01-03 by mbr0wn

muelthom, 2017-04-19:
I'm on a macOS machine and I've got the same issue by now:

```
mus@orion ~ $ pybombs prefix init ~/prefix -a myprefix -R gnuradio-default
PyBOMBS - INFO - PyBOMBS Version 2.3.0
Alias `myprefix' already exists, overwrite Y/[N]? Y
PyBOMBS.ConfigManager - INFO - Creating new config file /Users/mus/prefix/.pybombs/config.yml
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO - 
  - gnuradio
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
PyBOMBS.DepManager - ERROR - Package has no install method: ssl (declared as dependency for package libevent)
```

Checking for the recipe returns this:
```
mus@orion ~ $ pybombs recipes list -l ssl
PyBOMBS - INFO - PyBOMBS Version 2.3.0
Loading package information.....
Package Name  Recipe Filename                                Installed By  
---------------------------------------------------------------------------
libpolarssl   ~/.pybombs/recipes/gr-recipes/libpolarssl.lwr  -             
ssl           ~/.pybombs/recipes/gr-recipes/ssl.lwr          pkgconfig
```

I've also had a folder "~/gnuradio/" from a previous gnuradio-build attempt but I deleted it as recommended.

Any suggestions?

saintofinternet, 2020-01-03:
i am stuck on the same issue now! 

anyone out there with a solution?