gnuradio/pybombs #420: PyBOMBS seems to assume `pip` is `pip2`

fake-name, 2017-01-25:
Basically, my system has `pip` -> `pip3`, `pip3`, and `pip2`. This is because I installed `pip via `get-pip.py` in python 2, and then python 3, so the available install of `pip` was from the python 3 install.

This seems to break pybombs in a slightly confusing manner, as it doesn't explicitly reference this at all in the error output:

```
PyBOMBS.install_manager - INFO - Installation successful.
PyBOMBS.install_manager - INFO - Installing package: uhd
Cloning:     (100%) [=====================================================================================================================================================================================]
Configuring: (100%) [=====================================================================================================================================================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
--
-- Configuring the python interpreter...
-- Python interpreter: /usr/bin/python
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Operating on master branch.
-- Using UHD Images Directory: OFF
--
-- Configuring Boost C++ Libraries...
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   chrono
--   date_time
--   filesystem
--   program_options
--   regex
--   system
--   unit_test_framework
--   serialization
--   thread
--   atomic
-- Boost include directories: /usr/include
-- Boost library directories: /usr/lib/x86_64-linux-gnu
-- Boost libraries: /usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_program_options.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libpthread.so
--
-- Python checking for Python version 2.7 or greater
-- Python checking for Python version 2.7 or greater - found
--
-- Python checking for Mako templates 0.4.2 or greater
-- Python checking for Mako templates 0.4.2 or greater - "import mako" failed
--
-- Python checking for requests 2.0 or greater
-- Python checking for requests 2.0 or greater - found
--
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 1
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = FALSE
CMake Error at cmake/Modules/UHDComponent.cmake:51 (MESSAGE):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:367 (LIBUHD_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/durr/prefix/src/uhd/host/build/CMakeFiles/CMakeOutput.log".
See also "/home/durr/prefix/src/uhd/host/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.
durr@durr:~⟫

```

Manually retargetting `pip` to point to `pip2` fixes the issue, but it'd probably be better to explicitly reference `pip2`, or call `pip` as `python2 -m pip`.

skoslowski, 2017-01-26:
+1 for the later

E3V3A, 2017-03-27:
Your system works as expected and should. `pip` is always pointing to latest `pipN` install, AFAIK. So yes the latter is preferred, but the qestion remain why it doesn't work with P3?

fake-name, 2017-03-27:
@E3V3A - Presumably because I don't think PyBOMBS supported py3k when I opened the issue (I don't know if it does yet either).

mbr0wn, 2017-07-09:
PyBOMBS has supported Py3k for a while, but that's not what matters. A *prefix* will be running either Py2k or Py3k, and then we need to run the appropriate pip version.

mbr0wn, 2018-06-17:
Think this is fixed now.

closed 2018-06-17 by mbr0wn

fake-name, 2018-06-17:
Cool!