gnuradio/pybombs #421: Fix Ubuntu 14.04.5 apt-get bug with flag '--force-yes'

dimitriblock, 2017-01-26:
With new OS, pybombs 2.3.0 cannot install packages via apt-get and returns code 100. Running the command manually, also fails but recommends to set flag '--force-yes'.

I worked added this flag into packagers/apt.py line 123. Finally, the command works as excpected.

mbr0wn, 2017-02-09:
Can you post the failing command? `--force-yes` is recommended *NOT* to be used, and we already use `-y`.

mbr0wn, 2017-02-09:
```
       --force-yes
           Force yes; this is a dangerous option that will cause apt to continue without prompting if it is doing something
           potentially harmful. It should not be used except in very special situations. Using force-yes can potentially
           destroy your system! Configuration Item: APT::Get::force-yes. This is deprecated and replaced by
           --allow-downgrades, --allow-remove-essential, --allow-change-held-packages in 1.1.
```

You see my reluctance to merge this.

dimitriblock, 2017-02-17:
It's fine

closed 2017-02-17 by dimitriblock