gnuradio/pybombs #422: How do I move the pybombs prefix?

SonnyRajagopalan, 2017-02-01:
Hello,

I installed gnuradio components using `pybombs` in my local `/home` directory, but now I want to move it another filesystem with more space. What is the best way to move the `prefix` directory now at `~/prefix` to say, `/mylocalhost/pybombs/`? I cannot move the directory using `mv`, I assume, as there are other places where this will matter, as seen in the file `~/prefix/setup_env.sh`. 

Is there a clean way to do this? I don't want to break my installation. I am doing this on Ubuntu x86_64/14.04.

Thanks!

mbr0wn, 2017-02-03:
There's a bunch of ways your prefix is tied to your system.

If you're moving to another location on the same machine, you'll have to sed or search and replace paths in setup_env.sh. You might also have to manually update `~/.pybombs/config.yml` and `<prefix>/.pybombs/config.yml`.

This could be a feature for future PyBOMBS versions.

mbr0wn, 2017-02-03:
Closing 'cause no issue related.

closed 2017-02-03 by mbr0wn

giladbeeri, 2017-10-24:
A use case where this feature is useful:

If I both want to:

1. Install the prefix in a folder that requires root (e.g., /opt)
1. Use a git source location for a repository that requires authentication (assuming I use SSH for git auth, SSH key is under ~/.ssh)

Then using `sudo pybombs prefix init ...` will fail when it runs `git clone ...` because git will try to fetch the SSH key from root's storage (but it doesn't exist there) and using `pybombs prefix init ...` will fail because the user can't write to `/opt`. The workaround will be using `pybombs prefix init ...` in a user directory and then moving it manually to the root-written folder. 

So a feature of `pybombs prefix move ...` would be useful.

giladbeeri, 2017-10-27:
@mbr0wn executing your instructions for editing the files didn't end up well for me, because after moving the prefix, UHD looked for images in a directory relative to the old prefix.

Grepping the prefix found hundreds of mentions of the old prefix.

Why doesn't they all use an env var like "PREFIX" and use relative paths to that env var (so a change in root dir is simply a change to an env var, and the prefix is agnostic to its location)?