gnuradio/pybombs #423: Come up with a smart way to make PyBOMBS work with Py2+3

mbr0wn, 2017-02-09:
Here's the main issue: When installing a Python dependency, PyBOMBS might invoke the system pkg manager (e.g. apt-get). If it does, it has no way of knowing if the user is using Python 2 or 3. It could install `python-mako`, but we really need `python3-mako`. We currently have no way of doing this correctly.

E3V3A, 2017-03-27:
Well I guess the problem is really that most people have both, and is not clear which one is more compatible with gnuradio and its many OOTs. Of course we could make unreliable hacks, such as testing for which `pipX -V` has the latest version and look for time stamps on its binaries.

sbmueller, 2017-10-17:
+1 experiencing this issue right now

mbr0wn, 2018-06-17:
After thinking about this for almost two years (ahem, not all the time...) I still haven't come up with a better idea. I'll be going down the route described by @sbmueller .

mbr0wn, 2018-06-17:
See 7f2d6dea5a49e2296bed2a7f31d25cae531da37c and 7320ba61837267f4c3a29210ed7599a442013496.

closed 2018-06-17 by mbr0wn