gnuradio/pybombs #424: Package uhd has no recipe file

mschilp, 2017-02-21:
I'm trying to use pybombs to pull down uhd:
`pybombs install uhd`

I get the following response
`PyBOMBS - INFO - PyBOMBS Version 2.3.0
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.get_recipe - ERROR - Error fetching recipe `uhd':
Package uhd has no recipe file!
`
there was a fix that was posted in a discussion from Jan16, that called for changing the config.yml using lines that were found at:
https://github.com/gnuradio/pybombs/blob/master/pybombs/skel/.pybombs/config.yml

but they are no longer there.  Is there a way that I can get these lines or be directed to where I can find the fix to this particular issue? I appreciate the time on this.

mbr0wn, 2017-02-21:
Do you have the recipes installed? What does

    $ pybombs recipes list-repos

show?

mschilp, 2017-02-21:
Martin, I currently show the following in my recipes folder:
autoconf.lwr
build-essential.lwr
git.lwr
python.lwr
automake.lwr
cmake.lwr
graphviz.lwr
xz.lwr
autotools-bootstrap.lwr
default_prefix.lwr
python-requests.lwr

I do not have any uhd.lwr, also I don't know where to find it.  Is that
even what I'm supposed to be looking for?

I appreciate your time with this.

Mike

mbr0wn, 2017-02-21:
Go follow the instructions inthe manual, and you'll be good to go.

https://github.com/gnuradio/pybombs/blob/master/README.md

closed 2017-02-21 by mbr0wn

mschilp, 2017-02-21:
Thanks, I'll do that.

mschilp, 2017-02-22:
Hey Martin, so from the directions, since I already have pybombs installed
I ran:

$ pybombs recipes add gr-recipes
git+https://github.com/gnuradio/gr-recipes.git
$ pybombs recipes add gr-etcetera
git+https://github.com/gnuradio/gr-etcetera.git

I get the response that it has cloned 100% then:
PyBOMBS.recipes - ERROR - Could not fetch recipes: Process returned value: 128

I have seen some threads where other people had similar issues but I
don't see any solutions for the issue. I apologize for the repeated
questions, but I'm very new to this.

Thank you for your time.

Michael

mbr0wn, 2017-02-22:
Use `-vv`

akshadabhosale, 2020-03-30:
how to install UHD in GNU radio 3.8?