gnuradio/pybombs #425: Package PyBOMBS as native distro packages

marcusmueller, 2017-03-14:
PyBOMBS is becoming a bit stabler, and it'd be very helpful if we could have people do 

`dnf install pybombs`, `yum install pybombs` …

instead of the pip dance.

I know, it sounds stupid, but it takes away one step from the installation process, and also, it makes it much easier to build things like [live images](https://github.com/marcusmueller/fedora-live-GR/issues/6); in other words, working on this for RPM/Fedora, but not on debian.

Also avoids things like the pip2 / pip3 clash.

mbr0wn, 2017-03-15:
Sounds good. Volunteers welcome!

marcusmueller, 2017-03-16:
Done for Fedora 23–26, Mageia and RH/Centos/EPEL 6&7:

https://copr.fedorainfracloud.org/coprs/marcusmueller/pybombs/

balister, 2017-03-16:
On 03/16/2017 11:44 AM, Marcus Müller wrote:
> PyBOMBS is becoming a bit stabler, and it'd be very helpful if we could have people do 
> 
> `dnf install pybombs`, `yum install pybombs` …
> 
> instead of the pip dance.
> 
> I know, it sounds stupid, but it takes away one step from the installation process, and also, it makes it much easier to build things like [live images](https://github.com/marcusmueller/fedora-live-GR/issues/6); in other words, working on this for RPM/Fedora, but not on debian.
> 
> Also avoids things like the pip2 / pip3 clash.
> 
> - [x] Redhatalikes
> - [ ] Debianoids
> - [ ] Archies
> - [ ] Gento -O13
> - [ ] OpenEmbedded Recipe (calling @balister on that – do we have something like that around somewhere from E3xx?)

This is absurd. Much better just to write recipes for the things.

Philip

> - [ ] MacPorts
>

marcusmueller, 2017-03-16:
Well, I wholeheartedly agree, but if we all agreed this was madness,
we'd let PyBOMBS die today and bury it deep. I know having a metapackage
manager on an embedded device is retarded, but installing GNU Radio OOTs
is what folks use the embedded devices we're talking about for, so I'm
willing to subject myself to a bit of madness.

In reality, PyBOMBS is there to compensate for the lack of skills and
manpower of OOT-devs to create and maintain a package/recipe/… for every
software package manager/image build system out there.

So in that spirit, since no-one wants to write a pybombs-to-OE recipe
converter… can I haz oe-pybombs?