gnuradio/pybombs #428: Gnu-radio installation on Ubuntu with Pybombs fails

asim800, 2017-03-20:
Hi,

I was able to install gnuradio with build-gnuradio script after making following changes to fix CMake error of undefined reference to boost::program_options. 
export CC=/usr/bin/gcc-6
export CXX=/usr/bin/g++-6

However, I'd like to install gnuradio using pybombs so that I can add recipes later and use OOT flow. 

I ran the following command
susudo pybombs prefix init /usr/local -a usrlocal
sudo pybombs config default_prefix /usr/local/gr
sudo pybombs -v recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
sudo pybombs -v recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
sudo pybombs install uhd

Here again I am getting another undefined reference to boost library. I couldn't find the GCC compiler version supported by pybombs. Is there some documentation regarding the version numbers supporting tool chain?

../lib/libuhd.so.3.11: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
../lib/libuhd.so.3.11: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
collect2: error: ld returned 1 exit status

Thankyou

Here's slightly longer error log:
...
PyBOMBS.install_manager - DEBUG - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.Fetcher.git - DEBUG - Requirements met.
PyBOMBS.Fetcher.git - DEBUG - Using url - https://github.com/EttusResearch/uhd.git
PyBOMBS._process_thread() - DEBUG - Executing command `git clone https://github.com/EttusResearch/uhd.git uhd -b master'
Cloning:     (100%) [===============================================================================================]
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 0
PyBOMBS.Inventory - DEBUG - Setting state to `fetched'
PyBOMBS.Inventory - DEBUG - Saving inventory to file /home/hercules/coding/gnur/uhd/host/lib/.pybombs/inventory.yml...
PyBOMBS.Requirer - DEBUG - Requiring packages on host system: []
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.PackageManager - DEBUG - apt is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - yumdnf is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pymod
PyBOMBS.PackageManager - DEBUG - pymod is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using binary packagers: ['pip', 'apt', 'yumdnf', 'pymod', 'pkgconfig', 'cmd']
PyBOMBS.install_manager - DEBUG - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.PackageManager - DEBUG - Checking if package build-essential is installable...
PyBOMBS.Packager.apt - DEBUG - Package build-essential has version 12.1 in repositories
PyBOMBS.PackageManager - DEBUG - Checking if package build-essential is installed...
PyBOMBS.Packager.apt - DEBUG - Package build-essential has version 12.1 installed
PyBOMBS.install_manager - DEBUG - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.Requirer - DEBUG - Requirements met.
PyBOMBS.Packager.source - DEBUG - Using build directory: /home/hercules/coding/gnur/uhd/host/lib/src/uhd/host/build
PyBOMBS.Packager.source - DEBUG - Configuring recipe uhd
PyBOMBS.Packager.source - DEBUG - Using vars - {'config_opt': '', 'builddir': '/home/hercules/coding/gnur/uhd/host/lib/src/uhd/host/build'}
PyBOMBS.Packager.source - DEBUG - In cwd - /home/hercules/coding/gnur/uhd/host/lib/src/uhd/host/build
PyBOMBS._process_thread() - DEBUG - Executing command `$ cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/home/hercules/coding/gnur/uhd/host/lib  -Wno-dev'
Configuring: (100%) [===============================================================================================]
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 0
PyBOMBS.Packager.source - DEBUG - Configure successful.
PyBOMBS.Inventory - DEBUG - Setting state to `configured'
PyBOMBS.Inventory - DEBUG - Saving inventory to file /home/hercules/coding/gnur/uhd/host/lib/.pybombs/inventory.yml...
PyBOMBS.Packager.source - DEBUG - Building recipe uhd
PyBOMBS.Packager.source - DEBUG - In cwd - /home/hercules/coding/gnur/uhd/host/lib/src/uhd/host/build
PyBOMBS._process_thread() - DEBUG - Executing command `$ make -j4'
Building:    (100%) [===============================================================================================]
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 2
PyBOMBS.Packager.source - DEBUG - Building recipe uhd
PyBOMBS.Packager.source - DEBUG - In cwd - /home/hercules/coding/gnur/uhd/host/lib/src/uhd/host/build
PyBOMBS._process_thread() - DEBUG - Executing command `$ make -j1'
[ 61%] Built target uhd
[ 62%] Linking CXX executable sync_to_gps
../lib/libuhd.so.3.11: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
../lib/libuhd.so.3.11: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/sync_to_gps.dir/build.make:107: recipe for target 'examples/sync_to_gps' failed
make[2]: *** [examples/sync_to_gps] Error 1
CMakeFiles/Makefile2:414: recipe for target 'examples/CMakeFiles/sync_to_gps.dir/all' failed
make[1]: *** [examples/CMakeFiles/sync_to_gps.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.
.

mbr0wn, 2017-03-26:
Which Ubuntu version were you using?

There's an ABI issue we know about, but this doesn't look like it's a PyBOMBS issue.

asim800, 2017-03-27:
Thanks for looking into it.

It turns out that I get this error when I used version 6 of gcc/g++ with pybombs and I needed to use gcc/g++ version 5 to get around this problem. 

One lingering question that I have was that I was able to get gnuradio/uhd installed with gcc/g++ version 6 using build-gnuradio script from (www.sbrac.org/files/build-gnuradio).

regards

closed 2017-03-27 by asim800

febs, 2017-10-05:
I think this should not be closed, since it requires a very specific user intervention at the Operating System level to work. Thanks.

curtcorum, 2018-06-06:
I concur with Fabio @febs 
I am having similar issues with pybombs installing gnuradio on Ubuntu 16.04, after trying a number of variations from #428 and #448 as well as other links.

My goal is to build and use gnuradio with uhd and controlport support and pybombs seems to be the way to do it. No easy combination of existing  ubuntu/deb packages seems to support it (due to thrift dependency?)

curt@curt--laptop:~$ gcc --version
gcc (Ubuntu 5.5.0-12ubuntu1~16.04) 5.5.0 20171010

curt@curt--laptop:~$ pybombs --version
2.3.2

As of now after purging all packages related to gnuradio and uhd and concentraing on gcc-5 I can get thrift and uhd to build. Gnuradio still will not build due to the error:

[ 14%] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/float_array_to_int.cc.o
In file included from /usr/include/c++/5/cstdint:35:0,
                 from /home/curt/pybombs_180605/src/gnuradio/gr-blocks/lib/float_array_to_int.cc:30:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \

Any ideas?

thanks,
Curt

Full output:

curt@curt--laptop:~$ nice pybombs -v rebuild gnuradio
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - Parsing config file failed (/etc/pybombs/config.yml).
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /home/curt/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: ordereddict([('git-cache', '/home/curt/.pybombs/gitcache'), ('elevate_pre_args', ['sudo', '-H']), ('makewidth', 4), ('packagers', 'pip,apt,pymod,pkgconfig,cmd'), ('default_prefix', 'myprefix')])
PyBOMBS.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/dist-packages/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/curt/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias `myprefix'.
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/home/curt/pybombs_180605)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/curt/pybombs_180605/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/curt/pybombs_180605/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/curt/pybombs_180605/src
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /home/curt/pybombs_180605/.pybombs/inventory.yml...
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/curt/pybombs_180605/.pybombs/recipes
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /home/curt/pybombs_180605/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/home/curt/pybombs_180605/.pybombs/recipes', '/home/curt/.pybombs/recipes/gr-recipes', '/home/curt/.pybombs/recipes/gr-etcetera', '/usr/local/lib/python2.7/dist-packages/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/curt/pybombs_180605/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/curt/pybombs_180605/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/curt/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/curt/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 189 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/curt/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/curt/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 97 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/dist-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/dist-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.rebuild - DEBUG - Initializing command class for command rebuild
PyBOMBS.rebuild - DEBUG - Install tree:
|
\- gnuradio
PyBOMBS.rebuild - INFO - Rebuilding package: gnuradio
PyBOMBS.Inventory - DEBUG - Setting state to `fetched'
PyBOMBS.Requirer - DEBUG - Requiring packages on host system: []
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
[... 90 lines cut ...]
     ^
/home/curt/pybombs_180605/src/gnuradio/gr-blocks/lib/float_array_to_int.cc:40:9: error: ‘r’ was not declared in this scope
     if (r < MIN_INT)
         ^
/home/curt/pybombs_180605/src/gnuradio/gr-blocks/lib/float_array_to_int.cc:40:13: error: ‘MIN_INT’ was not declared in this scope
[... cut ...]
make[1]: *** [gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
PyBOMBS.Packager.source - ERROR - Build failed.
PyBOMBS.rebuild - ERROR - Error rebuilding package gnuradio. Aborting.