gnuradio/pybombs #431: A more versatile setup_env.sh

jaredd, 2017-04-13:
Often I find that I install OOT modules to a directory separate from my base gnuradio install in order to keep my base GR install clean and make nuking my OOT easy if I need to completely rebuild and reinstall.  The setup_env.sh is almost appropriate for also adding the OOT paths to the environment to pull in those blocks and libs except it explicitly uses the prefix directory in its paths.  Below is a portion of an (annotated) alternative setup_env.sh that could be copied to the appropriate OOT directory and operate as desired.

```bash
#!/bin/bash

# path to this file
OWNPATH=$(dirname $(realpath ${BASH_SOURCE[0]}))

# activate virtualenv first, shouldn't GR paths precede VE paths?
# If we're in a Python virtualenv, activate that
if [ -r $OWNPATH/bin/activate ]; then
    source $OWNPATH/bin/activate
fi

# change the IFS to do splitting / assembly with colons, save old IFS
OIFS=$IFS
export IFS=':'
# use path globs which are clearer than giant paragraph of paths
paths=($OWNPATH/{lib,lib64}/python{2.6,2.7}/{site,dist}-packages $PYTHONPATH)
# the following expands with colons b/w elements b/c IFS=':'
PYTHONPATH="${paths[*]}"
#
# repeat for other env vars
#
# restore IFS
export IFS=$OIFS

export PATH PYTHONPATH LD_LIBRARY_PATH LIBRARY_PATH PKG_CONFIG_PATH PYBOMBS_PREFIX
```

mbr0wn, 2017-07-09:
This looks like it's not portable to some other shells, but I like some of the concepts.

jaredd, 2017-07-10:
For this to truly work there needs to be some extra support for environment variable management from the virtualenv scripts.  I'll see what I can do there.  In the meantime, why can't there be several templates for scripts for different shells?  I feel like this is something that won't have to be updated too often (famous last words).  The setup-env scripts will also need an "undo" feature so one can switch b/w different OOT mixins through the environment.

giladbeeri, 2017-10-27:
This will also help with moving prefixes, see #422