gnuradio/pybombs #442: Installation  error!

jjbaek35, 2017-06-16:
Hi~all.

I'm struggling with following error!
Could you help me figure out what the problem is?
I install Ubuntu15.04 based Vmware workstation with assigned  20 GB HDD.
You can the error message at the end of logs.

---------------------------------------------------------------------------------------------------------------- 
root@ubuntu:/usr/local/bin# sudo pybombs install gr-gsm
PyBOMBS - INFO - PyBOMBS Version 2.3.0
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  python-colorama python-distlib python-requests python-wheel
Use 'apt-get autoremove' to remove them.
Suggested packages:
  libasound2-doc
The following NEW packages will be installed:
  libasound2-dev
0 upgraded, 1 newly installed, 0 to remove and 198 not upgraded.

Need to get 55.6 kB of archives.
After this operation, 445 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu/ vivid/main libasound2-dev i386 1.0.28-1 [55.6 kB]
Fetched 55.6 kB in 0s (102 kB/s)    

Selecting previously unselected package libasound2-dev:i386.
(Reading database ... 224309 files and directories currently installed.)
Preparing to unpack .../libasound2-dev_1.0.28-1_i386.deb ...
Unpacking libasound2-dev:i386 (1.0.28-1) ...
Setting up libasound2-dev:i386 (1.0.28-1) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done

The following packages were automatically installed and are no longer required:
  python-colorama python-distlib python-requests python-wheel
Use 'apt-get autoremove' to remove them.
Suggested packages:
  python-markdown python-pygments python-memcache
The following NEW packages will be installed:
  python-cheetah
0 upgraded, 1 newly installed, 0 to remove and 198 not upgraded.
Need to get 144 kB of archives.
After this operation, 788 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu/ vivid/main python-cheetah i386 2.4.4-3.fakesyncbuild1 [144 kB]
Fetched 144 kB in 0s (215 kB/s)    
Selecting previously unselected package python-cheetah.
(Reading database ... 224355 files and directories currently installed.)
Preparing to unpack .../python-cheetah_2.4.4-3.fakesyncbuild1_i386.deb ...
Unpacking python-cheetah (2.4.4-3.fakesyncbuild1) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up python-cheetah (2.4.4-3.fakesyncbuild1) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  python-colorama python-distlib python-requests python-wheel
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  libwxbase2.8-0 libwxgtk-media2.8-0 libwxgtk2.8-0 python-wxversion
Suggested packages:
  wx2.8-doc wx2.8-examples editra
The following NEW packages will be installed:
  libwxbase2.8-0 libwxgtk-media2.8-0 libwxgtk2.8-0 python-wxgtk2.8
  python-wxversion
0 upgraded, 5 newly installed, 0 to remove and 198 not upgraded.
[... 1211 lines cut ...]
Linking CXX executable usrp_burn_db_eeprom
[ 94%] Built target usrp_burn_db_eeprom
Scanning dependencies of target usrp_burn_mb_eeprom
[ 94%] Building CXX object utils/CMakeFiles/usrp_burn_mb_eeprom.dir/usrp_burn_mb_eeprom.cpp.o
Linking CXX executable usrp_burn_mb_eeprom
[ 94%] Built target usrp_burn_mb_eeprom
Scanning dependencies of target usrp_n2xx_simple_net_burner
[ 95%] Building CXX object utils/CMakeFiles/usrp_n2xx_simple_net_burner.dir/usrp_n2xx_simple_net_burner.cpp.o
Linking CXX executable usrp_n2xx_simple_net_burner
[ 95%] Built target usrp_n2xx_simple_net_burner
Scanning dependencies of target usrp_x3xx_fpga_burner
[ 95%] Building CXX object utils/CMakeFiles/usrp_x3xx_fpga_burner.dir/usrp_x3xx_fpga_burner.cpp.o
[ 95%] Building C object utils/CMakeFiles/usrp_x3xx_fpga_burner.dir/__/lib/usrp/x300/cdecode.c.o
Linking CXX executable usrp_x3xx_fpga_burner
[ 95%] Built target usrp_x3xx_fpga_burner
[ 95%] Built target doxygen_docs
[100%] Built target man_page_gzips
Installing:  (100%) [=========================================================]]
PyBOMBS.install_manager - INFO - Installation successful.
PyBOMBS.install_manager - INFO - Installing package: gnuradio
Cloning:     (100%) [=========================================================]
**PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Unable to fetch recipe gnuradio
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.**
root@ubuntu:/usr/local/bin#

mbr0wn, 2017-06-17:
Don't run this as root if you can avoid it. I guess in a VM, it's maybe OK.

I'm not sure what's wrong though. Clearly, cloning GNU Radio is failing, but why, I'm not sure.

jjbaek35, 2017-06-30:
I solved the errors by myself.

closed 2017-06-30 by jjbaek35

shawnerz, 2017-07-16:
jjbael35: Posting what you did to solve the problem would be helpful to the rest of the community.
Thank you.

tenoritaiga, 2017-07-23:
I'm running into the same issue. @jjbaek35 could you please explain how you fixed this?

tenoritaiga, 2017-07-23:
I think I may have more or less worked around this - at least, I can get a successful build.

Initialize like this:

`pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git`
`pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git`

Then I ran an init with extra verbosity to show any errors:

`pybombs prefix init ~/prefix -a myprefix -R gnuradio-default -vv`

This seemed to avoid the above error, but then I got a different one about VOLK not being available. I guess this is because the recipe adds above don't add the `--recursive` flag to the git clones, which is necessary to pull VOLK. So I just did it manually:

`cd prefix`
`rm -rf gnuradio`
`git clone --recursive https://github.com/gnuradio/gnuradio.git`

Then run the `prefix init` command above again, overwrite your existing profile, and you should get a successful build. I'm not sure why this worked but I can successfully run gnuradio-companion now. YMMV.

shawnerz, 2017-07-25:
Thanks for posting. I ending up overwriting my Ubuntu partition, starting from a clean install, installing Pybombs, installing the recipes, and then running the Gnuradio pybomb.
And as always with computers, YMMV. :)
-Shawn