gnuradio/pybombs #445: PyBOMBS-induced pip --system installation of python2-requests/PyYAML/… breaks updateability

marcusmueller, 2017-06-26:
If using PyBOMBS in its default config, it will use pip in system mode as package source

It shouldn't. I've just encountered the case where this seriously broke a whole lot of Python packages that, through upgrades, suddenly depended on distro python-packages, which PyBOMBS installed before via pip (although these could have been installed as distro packages).

Since that is against the PyBOMBS philosophy of not polluting the user's global installation, we should not allow pip to be the primary package source.

mbr0wn, 2017-07-09:
I had some good reason to prefer pip over system's packagers, but I'm not sure what it was. We still need to support PyBOMBS prefixes being virtualenvs -- maybe that was it.

awalls-cx18, 2017-08-01:
I always remove pip from the allowed package managers before I have pybombs install anything, just for this reason.

mbr0wn, 2017-11-03:
pip was demoted, and we don't even use PyYAML any more, so I'm closing this.

closed 2017-11-03 by mbr0wn